openthread/openthread #3641: SPI doesn't seem to be working with OTBR image

ntavish, 2019-03-01:
**Description**
Ive connected an nrf52840 board, with the NCP firmware compiled with SPI as the transport, on the correct pins for both nrf52840, and rpi3. I also had to slow the clock of the raspberrypi to for my saleae to capture the data.

[image]

**To Reproduce**
1. Compile and flash `ot-ncp-ftd` compiled with `make -f examples/Makefile-nrf52840 NCP_SPI=1`
2. Connect nrf52840 to rpi

| pin name |  nrf52840 pin | Rpi pin |
|--------------|-----------------|-----------|
| MOSI | 4 | GPIO10|
|MISO | 28 | GPIO9 | 
|SCK|3 | GPIO11|
|CSN | 29 | GPIO7|
|IRQ| 30| GPIO25|

3. Run docker image with `docker run --sysctl "net.ipv6.conf.all.disable_ipv6=0 net.ipv4.conf.all.forwarding=1 net.ipv6.conf.all.forwarding=1" -p 8080:80 --dns=127.0.0.1 -it --volume /dev/ttyACM0:/dev/ttyACM0 --privileged openthread/otbr --ncp-path /dev/spidev0.1 `

**Expected behavior**
`wpantund` should not end in error.

**Console/log output**
```
pi@raspberrypi:~ $ docker run --sysctl "net.ipv6.conf.all.disable_ipv6=0 net.ipv4.conf.all.forwarding=1 net.ipv6.conf.all.forwarding=1" -p 8080:80 --dns=127.0.0.1 -it --volume /dev/ttyACM0:/dev/ttyACM0 --privileged openthread/otbr --ncp-path /dev/spidev0.1 
WARNING: Localhost DNS setting (--dns=127.0.0.1) may fail in containers.
--ncp-path
NCP_PATH: /dev/spidev0.1
TUN_INTERFACE_NAME: wpan0
NAT64_PREFIX: 64:ff9b::/96
AUTO_PREFIX_ROUTE: true
AUTO_PREFIX_SLAAC: true
Current platform is ubuntu
* Applying /etc/sysctl.d/10-console-messages.conf ...
kernel.printk = 4 4 1 7
* Applying /etc/sysctl.d/10-ipv6-privacy.conf ...
net.ipv6.conf.all.use_tempaddr = 2
net.ipv6.conf.default.use_tempaddr = 2
* Applying /etc/sysctl.d/10-kernel-hardening.conf ...
kernel.kptr_restrict = 1
* Applying /etc/sysctl.d/10-link-restrictions.conf ...
fs.protected_hardlinks = 1
fs.protected_symlinks = 1
* Applying /etc/sysctl.d/10-magic-sysrq.conf ...
kernel.sysrq = 176
* Applying /etc/sysctl.d/10-network-security.conf ...
net.ipv4.conf.default.rp_filter = 1
net.ipv4.conf.all.rp_filter = 1
net.ipv4.tcp_syncookies = 1
* Applying /etc/sysctl.d/10-ptrace.conf ...
* Applying /etc/sysctl.d/10-zeropage.conf ...
vm.mmap_min_addr = 32768
* Applying /etc/sysctl.d/60-otbr-ip-forward.conf ...
net.ipv6.conf.all.forwarding = 1
net.ipv4.ip_forward = 1
* Applying /etc/sysctl.conf ...
 * Starting userspace NAT64 tayga                                                                                                                                     [ OK ] 
/usr/sbin/service
 * Starting domain name service... bind9                                                                                                                              [ OK ] 
/usr/sbin/service
 * dbus is not running
 * Starting system message bus dbus                                                                                                                                   [ OK ] 
   ...fail!
wpantund[132]: Starting wpantund 0.08.00d (Nov 21 2018 21:32:05) . . .
wpantund[132]: 	BUILD_VERSION = a1e0dfe
wpantund[132]: Configuration file "/etc/wpantund.conf" read.
wpantund[132]: Ready. Using DBUS bus ":1.0"
wpantund[132]: Running as root without dropping privileges!
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
   ...fail!
   ...fail!
otWeb[148]: border router web started on wpan0
otWeb[148]: message is NULL
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
otbr-agent[163]: Starting border router agent on wpan0...
otbr-agent[163]: NCP requesting DBus name otbr.agent.wpan0...
otbr-agent[163]: NCP failed to find the interface!
otbr-agent[163]: NCP failed to initialize!
otbr-agent[163]: Failed to create border route agent instance: -1!
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: SpinelNCPInstance-Protothreads.cpp:412: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: Resetting and trying again... (retry 1)
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: Resetting and trying again... (retry 2)
wpantund[132]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: Resetting and trying again... (retry 3)
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: SpinelNCPInstance-Protothreads.cpp:400: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: SpinelNCPInstance-DataPump.cpp:239: Requirement Failed (mInboundFrameSize < sizeof(mInboundFrame))
wpantund[132]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[132]: State change: "uninitialized" -> "uninitialized:fault"
``

ntavish, 2019-03-01:
Attaching Saleae logic analyzer exported file also.

https://drive.google.com/file/d/1hNA-SW0ZhLQYMxe7AuXh3eHYGlvkfiWK/view?usp=sharing

LuDuda, 2019-03-01:
Hello @ntavish 
At first glance, it seems that you have an incorrect [path](https://github.com/openthread/borderrouter/blob/82d4682d7e080437eaaa94c7f6951e75c0681caa/script/docker_entrypoint.sh#L87) to NCP device in `wpantund.conf` file.

As [README.md](https://github.com/openthread/openthread/tree/master/examples/platforms/nrf52840) states, the correct value of `Config:NCP:SocketPath` variable for your case (`spidev0.1` used) is:

`Config:NCP:SocketPath "system:/usr/bin/spi-hdlc-adapter --gpio-int /sys/class/gpio/gpio25 /dev/spidev0.1"`

(of course with the assumption that `spi-hdlc-adapter` is located in the `/usr/bin` directory).

We have recently checked SPI communication with NCP and RCP on RPi 3B with the latest OpenThread libraries, so it should work fine. Please let me know about the progress.

ntavish, 2019-03-02:
Hi @LuDuda, thanks for the response, that solved the problem for me, not sure how I missed that!

Though one problem still persists for me, half the times wpantund fails to initialize, and sometimes when it does, it fails:

```
spi-hdlc-adapter[137]: INFO: sHdlcRxBadCrcCount=0
wpantund[132]: [-NCP-]: NCP was reset (RESET_SOFTWARE, 114)
wpantund[132]: [-NCP-]: NCP was reset (RESET_SOFTWARE, 114)
wpantund[132]: Unexpected reset during NCP initialization.
wpantund[132]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[132]: State change: "uninitialized" -> "uninitialized:fault"
```
It might be because the clock speed for SPI is too high, it seems to be 125MHz in the raspian image I'm using:

```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.8 (stretch)
Release:	9.8
Codename:	stretch
```

The workaround for me has been to reduce default clock speed by loading a device tree overlay to slow down the clock before running the docker image. Listing steps in case someone else needs to do this also.

File: `slowspi-overlay.dts`
```
/dts-v1/;
/plugin/;
 
/ {
   compatible = "brcm,bcm2835", "brcm,bcm2836", "brcm,bcm2708", "brcm,bcm2709";
 
   fragment@0 {
      target = <&spi0>;
      frag0: __overlay__ {
         #address-cells = <1>;
         #size-cells = <0>;
         status = "okay";
         cs-gpios = <&gpio 8 1>, <&gpio 7 1>;
 
         spidev@0{
            compatible = "spidev";
            reg = <0>;   /* CE0 */
            #address-cells = <1>;
            #size-cells = <0>;
            spi-max-frequency = <100000>;
         };
 
         spidev@1{
            compatible = "spidev";
            reg = <1>;   /* CE1 */
            #address-cells = <1>;
            #size-cells = <0>;
            spi-max-frequency = <100000>;
         };
      };
   };
 
   fragment@1 {
      target = <&spi0_cs_pins>;
      frag1: __overlay__ {
        brcm,pins = <8 7>;
        brcm,function = <1>; /* out */
      };
   };
};
```

Compile it with `dtc -O dtb -o slowspi-overlay.dtbo -b 0 -@ slowspi-overlay.dts`

Load it with `sudo dtoverlay slowspi-overlay`. (If you need to remove it later, run `sudo dtoverlay -r slowspi-overlay`).

LuDuda, 2019-03-05:
Hmm, I'm not familiar with such issue. @pszkotak any thoughts?

pszkotak, 2019-03-05:
@LuDuda this is a different case than ours. Let me explain.

@ntavish does not run the OTBR image that is provided with nRF5 SDK for Thread and Zigbee, but some other distribution with docker. The image that we do test and share to public is based on raspbian and native wpantund. The SPI clock is set to 1MHz.

I propose to update the configuration/documentation for the SPI to:

- for NCP:
`Config:NCP:SocketPath "system:/usr/bin/spi-hdlc-adapter --gpio-int /sys/class/gpio/gpio25 --spi-speed 1000000 /dev/spidev0.1"`
- for RCP:
`Config:NCP:SocketPath "system:/usr/bin/ot-ncp /usr/bin/spi-hdlc-adapter -- '--stdio -i /sys/class/gpio/gpio25 --spi-speed 1000000 /dev/spidev0.1'"`

@ntavish, could you tell us more details about the OS you run on the RPi? Is it Raspian, Ubuntu or anything else? We build custom Raspian ourselves - it defaults the SPI clock to 1MHz.

ntavish, 2019-03-05:
@pszkotak I'm running Raspbian too, Raspbian 9.8 stretch

```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.8 (stretch)
Release:        9.8
Codename:       stretch
```

Default clock on this seems to be 125MHz on this. I don't think I might have changed this, just enabled SPI (using raspi-config) and checked the clock speed. I checked it on another pi with release 9.6, clock seems to be 125MHz on that also.

```
pi@prod:~ $ cat /sys/class/spidev/spidev0.1/device/of_node/spi-max-frequency | od -t d --endian=big
0000000   125000000
0000004
pi@prod:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.6 (stretch)
Release:        9.6
Codename:       stretch
```

pszkotak, 2019-03-05:
@ntavish Thank you for a quick response.
I will check our build settings then, maybe we have some old code everyone forgot about that sets this ;).

Could you check if the configuration I posted (`Config:NCP:SocketPath "system:/usr/bin/spi-hdlc-adapter --gpio-int /sys/class/gpio/gpio25 --spi-speed 1000000 /dev/spidev0.1"`) can be applied to your wpantund.conf in the docker environement? It should solve the issue without recompiling the dts.

ntavish, 2019-03-05:
That doesn't seem to be working for me. `wpantund` fails to initialize. Lower clocks are working though, tested with 700KHz and 500KHz.

    "system:/usr/bin/spi-hdlc-adapter --gpio-int /sys/class/gpio/gpio25 --spi-speed 500000 /dev/spidev0.1"

closed 2019-03-12 by ntavish