openthread/openthread #3642: Questions about the EFR32 radio driver

dismirlian, 2019-03-03:
Hi all, I have some questions about the EFR32 radio driver. I have been analyzing the code while waiting for some test hardware to arrive, so these are just questions, not confirmed bugs. In any case, some of the code looks suspicious. Sorry if some of these are dumb comments or questions.

1. According to the IEEE802.15.4 spec, the ACK timeouts are:
   * macAckWaitDuration = aUnitBackoffPeriod + aTurnaroundTime + phySHRDuration + 6 × phySymbolsPerOctet = 54 symbols = 864us, in case of beacon-enabled PANs
and
   * macAckWaitDuration = aTurnaroundTime + phySHRDuration + 6 × phySymbolsPerOctet = 34 symbols = 544us, in case of non-beacon-enabled PANs

   Is there any particular reason for the ACK timeout to be set to 894us instead of 864us?
https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L145-L150

2. `RAIL_InitTxPowerCurves` and `RAIL_ConfigTxPower` are called each time `efr32RadioSetTxPower` is called. These functions initialize the translation curves from dBm to some magic numbers, and configure the PAs, which I understand should be initialized only once (during initialization). However, `otPlatRadioTransmit` calls `efr32RadioSetTxPower` each time the band changes, which seems unnecessary. In contrast, `otPlatRadioSetTransmitPower` only calls `RAIL_SetTxPowerDbm`, which (I think) is the only necessary call to change the TX power.

3. All the scheduler stuff in multiple functions (`*scheduler*`) is only necessary if using the multiprotocol version of the RAIL library. The linked version of the library is not the multiprotocol version, so all these scheduler stuff seems to be unused, and therefore could be removed. I agree that this could be useful in some dynamic multiprotocol scenarios (eg. OpenThread + BLE), but this could be a compile option. I don't know how hard this is, because the build system should link either the multiprotocol or the standard version of RAIL.

4. (This is not specific to the EFR32 implementation) I think there is contradicting information (or maybe I am misunderstanding):
   * According to the state transition diagram in <openthread/platform/radio.h>, TRANSMIT ==> RECEIVE is a valid state transition.
https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/include/openthread/platform/radio.h#L216-L225
   * According to the same header, `otPlatRadioReceive`'s return value documentation states:  
     > @retval OT_ERROR_INVALID_STATE The radio was disabled or transmitting.

   Does the state diagram mean that a `otPlatRadioReceive` can abort an ongoing transmit?

5. `otPlatRadioReceive` can return `OT_ERROR_INVALID_ARGS` and `OT_ERROR_FAILED`, but these are not listed among the possible return values of this function in <openthread/platform/radio.h>. As a general question: are *other* return values possible, apart from those stated in the header? Are the callers designed to fail "gracefully" due to generic/unhandled return errors?

6. `otPlatRadioTransmit` can return `OT_ERROR_INVALID_ARGS`, but this is not listed in the header file as one the possible return values for this function.

7. If the band changed from the previous transmit to the current one, `otPlatRadioTransmit` calls `efr32RadioSetTxPower` to set the current band's power. I can't find in the RAIL docs any indication that this would be necessary; each RAIL instance should maintain its own output power configuration.

8. `otPlatRadioGetCaps` indicates that the EFR32 radio has the following capabilities: `OT_RADIO_CAPS_ACK_TIMEOUT` and `OT_RADIO_CAPS_CSMA_BACKOFF`. Shouldn't we include `OT_RADIO_CAPS_TRANSMIT_RETRIES`?

9. I think the following should be `static`: `findSrcMatchAvailEntry`, `findSrcMatchShortEntry`, `findSrcMatchExtEntry`, `addToSrcMatchShortIndirect`, `addToSrcMatchExtIndirect`, `removeFromSrcMatchShortIndirect`, `removeFromSrcMatchExtIndirect`, etc.

10. `removeFromSrcMatchExtIndirect`/`removeFromSrcMatchShortIndirect` use `memset` to clear the uint16_t checksum. Why? It would be simpler just to set `table[entry].checksum = 0;`.
https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L713-L716

11. Why is the source match entry table not storing the whole addresses and only the checksums? Is this safe?

12. In `processNextRxPacket`, `memcpy` is used to copy the packet. In the second call to `memcpy`, `lastPortionData` can be `NULL` (but fortunately the # of bytes to copy will be 0). I don't think this will break `memcpy`, but I don't know if it is standard compliant. The RAIL function `RAIL_CopyRxPacket` deals with this case, and may replace both `memcpy`s.
https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L856-L859

13. In `processNextRxPacket`,
https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L869-L872
the code checks `packetDetails.isAck`, and then asserts if the received packet doesn't have a length of `IEEE802154_ACK_LENGTH` (=5). However, RAIL documentation for the `isAck` field states:
    > isAck: Indicates whether the received packet is an ACK. An 'ACK' is defined as a packet received during the RX ACK window when auto ACK is enabled. Set to 0 if not an ACK, and 1 for an ACK. It is always 0 if auto ACK is not enabled.

    So, if a non-ack packet is received during the RX ACK window, it will be flagged "isAck" even if it is not really an ACK. I think the code should not `assert`, but `otEXPECT`, and possibly flag an ACK timeout to OT core.

14. `processNextRxPacket` checks these two conditions:
https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L842-L846
I think the first should never happen in normal operation (indicates a bug in RAIL). Regarding the second, the EFR32 radio configured in IEEE802.15.4 mode should already filter the packets with length <5 and >127. Is there a reason why these are not `assert`s? Are they some sort of workaround for RAIL bugs?

15. In the following lines:
https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L887-L907
`processNextRxPacket` first discards any packet with `length != IEEE802154_ACK_LENGTH` (actually, this block is only hit if `length >= IEEE802154_MIN_LENGTH && length <= IEEE802154_MAX_LENGTH`. So I think line 902 can be simplified to `if (sPromiscuous)`.

16. In the Event handler callback, a new RX packet is always held:
https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L964-L967
However, the RAIL sample application (railtest), does the following upon receiving a packet:
    ``` C
     // If we have just received an ACK, don't respond with an ACK
    if (rxPacket->dataPtr[2] == 0xF1) {
        RAIL_CancelAutoAck(railHandle);
    }
    ```
    This code is there because, if autoAck is enabled, the EFR32 will ACK a received ACK. I don't know if this was already taken into account in the EFR32 OT radio driver. Unfortunately, `RAIL_CancelAutoAck` should be called immediately after the packet reception (during the Rx->Tx turnaround time), before it's too late to cancel the outgoing ACK. I think it can't be delayed to `processNextRxPacket`.

17. If calibration is needed, the event callback calls `RAIL_Calibrate`, which can take several hundreds of milliseconds.
https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L993-L997
I know that a calibration request happens sporadically, but I think this call should be moved out of interrupt context.

Thank you!
Diego.

marvensilabs, 2019-03-20:
Hi Diego,  I'll do my best to answer your questions below regarding the EFR32 target.  Some questions may be better answered by the OpenThread team.

> 1. According to the IEEE802.15.4 spec, the ACK timeouts are:
>
>    * macAckWaitDuration = aUnitBackoffPeriod + aTurnaroundTime + phySHRDuration + 6 × phySymbolsPerOctet = 54 symbols = 864us, in [...]

Seems like a typo. For now, we would recommend changing to 864 until we have more clarification in our documentation and possibly a new number to recommend. The code looks like it could support sub-GHz but if so we are surprised the same value is used for 2.4GHz and 915MHz since we base this value off the symbol rate and our symbol rate is not the same on both PHYs.

> 2. `RAIL_InitTxPowerCurves` and `RAIL_ConfigTxPower` are called each time `efr32RadioSetTxPower` is called. These functions initialize the translation curves from dBm to some magic numbers, and [...]

RAIL_InitTxPowerCurves() sets up all power curves for all bands and should only need to be called once during initialization unless you expect these to change for your board at runtime. As for RAIL_ConfigTxPower(), you do need to call this when switching between bands on the same RAIL handle since you will need to switch the PA in use. That being said, it looks like you guys are using multiple RAIL handles and from answers above it sounds like you’re using the single protocol RAIL library. This is not something we support though. So if this file is being built with both RADIO_SUPPORT_915MHZ_OQPSK and RADIO_SUPPORT_2P4GHZ_OQPSK defined but using the non-multiprotocol library there may be some bugs where we are assuming each handle has full access to the hardware. So while I think this somewhat answers the question above, in general, I’m worried that the current dual mode implementation is not valid.

> 3. All the scheduler stuff in multiple functions (`*scheduler*`) is only necessary if using the multiprotocol version of the RAIL library. The linked version of the library is not the multiprotocol [...]
 
The SchedulerInfo_t parameter to RAIL API functions is only needed for multiprotocol so be can set to NULL and the structs removed.

 
> 4. (This is not specific to the EFR32 implementation) I think there is contradicting information (or maybe I am misunderstanding):
>    
>    * According to the state transition diagram in <openthread/platform/radio.h>, TRANSMIT ==> RECEIVE is a valid state transition.
>      
>        
>          
>            [openthread/include/openthread/platform/radio.h](https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/include/openthread/platform/radio.h#L216-L225)
>          
>          
>              Lines 216 to 225
>            in
>            [bf794ca](/openthread/openthread/commit/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999)
>          
>                  * The following are valid radio state transitions: 
>                  * 
>                  *                                    (Radio ON) 
>                  *  +----------+  Enable()  +-------+  Receive() +---------+   Transmit()  +----------+ 
>                  *  |          |----------->|       |----------->|         |-------------->|          | 
>                  *  | Disabled |            | Sleep |            | Receive |               | Transmit | 
>                  *  |          |<-----------|       |<-----------|         |<--------------|          | 
>                  *  +----------+  Disable() +-------+   Sleep()  +---------+   Receive()   +----------+ 
>                  *                                    (Radio OFF)                 or 
>                  *                                                        signal TransmitDone
>    * According to the same header, `otPlatRadioReceive`'s return value documentation states:
>      > @RetVal OT_ERROR_INVALID_STATE The radio was disabled or transmitting.
>    
>    Does the state diagram mean that a `otPlatRadioReceive` can abort an ongoing transmit?

I will leave it for the OpenThread team to answer.

> 5. `otPlatRadioReceive` can return `OT_ERROR_INVALID_ARGS` and `OT_ERROR_FAILED`, but these are not listed among the possible return values of this function in <openthread/platform/radio.h>. As a [...]

The allowed return values need clarifying by the OpenThread team. It should be a trivial change to convert errors into OT_ERROR_INVALID_STATE if needed.
 
<openthread/platform/radio.h> lists these two return values for otPlatRadioReceive
OT_ERROR_NONE          Successfully transitioned to Receive.
OT_ERROR_INVALID_STATE The radio was disabled or transmitting.
 
EFR32 code currently returns one of:
OT_ERROR_NONE
OT_ERROR_INVALID_STATE
OT_ERROR_INVALID_ARGS
OT_ERROR_FAILED
 
Most platforms stick to the OT_ERROR_NONE and OT_ERRROR_INVALID_STATE values but some return other values such as:
OT_ERROR_NONE
OT_ERROR_BUSY
OT_ERROR_FAILED
 
Code that directly calls otPlatRadioReceive asserts or tests for OT_ERROR_NONE.

> 6. `otPlatRadioTransmit` can return `OT_ERROR_INVALID_ARGS`, but this is not listed in the header file as one the possible return values for this function.

[... 167 lines cut ...]
>     }
>     ```
>     
>     
>     This code is there because, if autoAck is enabled, the EFR32 will ACK a received ACK. I don't know if this was already taken into account in the EFR32 OT radio driver. Unfortunately, `RAIL_CancelAutoAck` should be called immediately after the packet reception (during the Rx->Tx turnaroun [... cut ...]
Whilst IR Calibration can take 200ms for sub-GHz for 2.4GHz the data is stored in the Device Information page at production time. The API itself is safe to call from interrupt context.  I guess it depends on how important sub-GHz support is. We store the IRCAL value in RAM after the first run so only the first call to this would take any significant time. IRCALs also aren’t triggered at random, they are only run when you switch to a new channel configuration and there’s only one sub gig channel configuration so this will only happen one time after bootup. TEMPCAL takes basically no time in the API call (it triggers a radio operation that takes ~100us) so it should always be safe.

dismirlian, 2019-03-20:
Hi Marven, many thanks for your detailed answer.
> Seems like a typo. For now, we would recommend changing to 864 until we have more clarification in our documentation and possibly a new number to recommend. The code looks like it could support [...]
Ok, so 894us should be changed to 864us. However, I think these values would be correct for 915MHz also, because the IEEE 802.15.4-2011 standard says:

> 10.3.3 Symbol rate
> The O-QPSK PHY symbol rate shall be 25 ksymbol/s when operating in the 868 MHz band and 62.5 ksymbol/s when operating in the 780 MHz, 915 MHz, or 2450 MHz band with an accuracy of ±40 ppm.

A change would be necessary for the 868MHz band (which the code doesn't support yet).

> RAIL_InitTxPowerCurves() sets up all power curves for all bands and should only need to be called once during initialization unless you expect these to change for your board at runtime.

Got it. So the code could be simplified.

> As for RAIL_ConfigTxPower(), you do need to call this when switching between bands on the same RAIL handle since you will need to switch the PA in use. That being said, it looks like you guys are [...]

Ok, so, if I understand correctly, you are saying that the multiprotocol RAIL library is required *even if there is only one band in use at a given moment*? The current code links in the "non-multiprotocol" version of RAIL, so maybe the guy who wrote the original code worked around this by calling RAIL_ConfigTxPower when switching bands, as you suggest.

Just to clarify: given that OT uses only one band at a time, would it be safe to leave the code as-is, and linking the non-multiprotocol RAIL library?

> The SchedulerInfo_t parameter to RAIL API functions is only needed for multiprotocol so be can set to NULL and the structs removed.

Ok, this could be done if *either* 915MHz *or* 2.4GHz is needed, as per your answer above.

> I will leave it for the OpenThread team to answer.

Great. Maybe I should have opened another question with the non-platform-specific questions...

> The allowed return values need clarifying by the OpenThread team. It should be a trivial change to convert errors into OT_ERROR_INVALID_STATE if needed.
>
> <openthread/platform/radio.h> lists these [...]

It's clear, thanks.

> Again the allowed return values need clarifying by OpenThread team.
>
> <openthread/platform/radio.h> defines the following return values for otPlatRadioTransmit
> OT_ERROR_NONE Successfully transitioned [...]

Ok.

> I don’t know what OT_RADIO_CAPS_TRANSMIT_RETRIES means exactly. Reading through the code I think it’s trying to count the 5 CCA attempts that we do though it would take more understanding of [...]

On a second read of the header, it seems that OT_RADIO_CAPS_CSMA_BACKOFF implies the 5 CCA attempts, and OT_RADIO_CAPS_TRANSMIT_RETRIES implies both the 5 CCA attempts *and* some kind of retransmision. This is confirmed by the following discussion:

https://github.com/openthread/openthread/pull/2401#discussion_r156225368

Here @jwhui implies that OT_RADIO_CAPS_TRANSMIT_RETRIES means that the platform layer implements automatic retransmissions on no ack, so IMHO the code is OK as is.

> They could be made static as no external code or tests call these functions directly.

Ok.

> It should just be "checksum = 0" instead of memset. I don't see a reason to use memset here.

Ok.

> I think this is a question for the OpenThread team.
>
> It’s not a part of the standard and is probably going to be prone to collisions (especially for long addresses). I guess the net effect is that [...]

Great; so we need more input from the OT team (or the original code contributor).

> RAIL_CopyRxPacket should be used instead of two memcpy calls.

I agree.

> 
> > 13. In `processNextRxPacket`,
> >    ```
> >      [openthread/examples/platforms/efr32/radio.c](https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L869-L872)
[... 11 lines cut ...]
> >    
> >    So, if a non-ack packet is received during the RX ACK window, it will be flagged "isAck" even if it is not really an ACK. I think the code should not `assert`, but `otEXPECT`, and possibly flag an ACK timeout to OT core.
> 
> The documentation being referred to needs a note that protocol-specific Auto ACKing is special so we will add that. In 15.4 we should only pass up a packet that has a length of 5, the frame type set to ACK, and the same sequence number as the transmitted packet. So it should be safe to assert in this code path. Some older versions of RAIL were not as good at the sequence number checking but it looks like they are handling that separately so it should be fine.

Ah! Ok, so the special Auto-ACKing is enabled by calling `RAIL_IEEE802154_Init`, right? In this case, the original code is ok. Could we remove the packet sequence number checking with RAIL 2.5?

> we agree that it would be a bug on our side if any of these conditions happened. We are always filtering packets less than the minimum and greater than the maximum in 15.4 mode.

Ok, so we can change those `otEXPECT`s to `assert`s.

> I think if it was simplified to “if (sPromiscuous)” this would prevent it calling otRadioReceiveDone when not in promiscuous mode. Perhaps the test could be removed and always call [...]

[... cut ...]
> Whilst IR Calibration can take 200ms for sub-GHz for 2.4GHz the data is stored in the Device Information page at production time. The API itself is safe to call from interrupt context. I guess it [...]

Ok, so basically this would be a problem in the sub-GHz band. I think it won't be too hard to move the calibration out of the IRQ handler. If I understand correctly, the calibration event signals that calibration should be done in the near future (it's not compulsory to do it in the callback).

Thanks!!
Diego.

jwhui, 2019-03-21:
@dismirlian, thanks for raising these questions. As always, contributions are always welcome!

Let me try to answer the ones that are independent of EFR32 and RAIL.

> 4. (This is not specific to the EFR32 implementation) I think there is contradicting information (or maybe I am misunderstanding):
>
>    * According to the state transition diagram in [...]

Yes, an `otPlatRadioReceive()` can abort an ongoing transmit operation. Note that waiting for an IEEE 802.15.4 ACK is part of the transmit state. If OpenThread is handling ACK timeouts, then it uses `otPlatRadioReceive()` to abort waiting for an ACK.

> 5. `otPlatRadioReceive` can return `OT_ERROR_INVALID_ARGS` and `OT_ERROR_FAILED`, but these are not listed among the possible return values of this function in <openthread/platform/radio.h>. As a [...]

> 6. `otPlatRadioTransmit` can return `OT_ERROR_INVALID_ARGS`, but this is not listed in the header file as one the possible return values for this function.

OpenThread is designed to follow the documented radio driver state machine. There is an `assert()` to check when `otPlatRadioTransmit()` returns anything other than `OT_ERROR_NONE`.

> 8. `otPlatRadioGetCaps` indicates that the EFR32 radio has the following capabilities: `OT_RADIO_CAPS_ACK_TIMEOUT` and `OT_RADIO_CAPS_CSMA_BACKOFF`. Shouldn't we include `OT_RADIO_CAPS_TRANSMIT_RETRIES`?

`OT_RADIO_CAPS_TRANSMIT_RETRIES` is for retransmissions in response to no ack received. The name/docs could probably be updated to reflect this.

> 9. I think the following should be `static`: `findSrcMatchAvailEntry`, `findSrcMatchShortEntry`, `findSrcMatchExtEntry`, `addToSrcMatchShortIndirect`, `addToSrcMatchExtIndirect`, `removeFromSrcMatchShortIndirect`, `removeFromSrcMatchExtIndirect`, etc.

Agree.

> 10. `removeFromSrcMatchExtIndirect`/`removeFromSrcMatchShortIndirect` use `memset` to clear the uint16_t checksum. Why? It would be simpler just to set `table[entry].checksum = 0;`. [...]

Agree.

> 11. Why is the source match entry table not storing the whole addresses and only the checksums? Is this safe?

Many IEEE 802.15.4 radios that implement auto-ack with frame pending in hardware use this trick. It allows fixed-sized, 2-byte entries to be used for both short and extended addresses.

Given that the EFR32 driver implements this in software, it may be fine to store the whole address. However, I think some consideration should be given to the cost of searching through the source match table given that there is a hard real-time requirement.

> 12. In `processNextRxPacket`, `memcpy` is used to copy the packet. In the second call to `memcpy`, `lastPortionData` can be `NULL` (but fortunately the # of bytes to copy will be 0). I don't think [...]

Agree.

> 14. `processNextRxPacket` checks these two conditions:
> https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L842-L846
> I think the [...]

We can change to asserts. The code here is an evolution from before RAIL provided an IEEE 802.15.4 layer.

> 15. In the following lines:
> https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L887-L907
> `processNextRxPacket` first discards any [...]

Agree.

dismirlian, 2019-03-21:
> @dismirlian, thanks for raising these questions. As always, contributions are always welcome!

Hi @jwhui, thanks for your answer. I'll be happy to contribute some improvements to the driver once I get the hardware to test the modifications (unfortunately, my boards are stuck in customs).

> 
> Let me try to answer the ones that are independent of EFR32 and RAIL.
> 
> > 4. (This is not specific to the EFR32 implementation) I think there is contradicting information (or maybe I am misunderstanding):
> >    
> >    * According to the state transition diagram in <openthread/platform/radio.h>, TRANSMIT ==> RECEIVE is a valid state transition.
> >            [openthread/include/openthread/platform/radio.h](https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/include/openthread/platform/radio.h#L216-L225)
> >              Lines 216 to 225
> >            in
> >            [bf794ca](/openthread/openthread/commit/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999)
> >                
> >                  * The following are valid radio state transitions: 
> >                  * 
> >                  *                                    (Radio ON) 
> >                  *  +----------+  Enable()  +-------+  Receive() +---------+   Transmit()  +----------+ 
> >                  *  |          |----------->|       |----------->|         |-------------->|          | 
> >                  *  | Disabled |            | Sleep |            | Receive |               | Transmit | 
> >                  *  |          |<-----------|       |<-----------|         |<--------------|          | 
> >                  *  +----------+  Disable() +-------+   Sleep()  +---------+   Receive()   +----------+ 
> >                  *                                    (Radio OFF)                 or 
> >                  *                                                        signal TransmitDone
> >    * According to the same header, `otPlatRadioReceive`'s return value documentation states:
> >      > @RetVal OT_ERROR_INVALID_STATE The radio was disabled or transmitting.
> >    
> >    Does the state diagram mean that a `otPlatRadioReceive` can abort an ongoing transmit?
> 
> Yes, an `otPlatRadioReceive()` can abort an ongoing transmit operation. Note that waiting for an IEEE 802.15.4 ACK is part of the transmit state. If OpenThread is handling ACK timeouts, then it uses `otPlatRadioReceive()` to abort waiting for an ACK.

Ok, so, if I understand correctly, this is the only case in which a receive can abort a transmit:
https://github.com/openthread/openthread/blob/e02c9702694ba6a1f15f25d7c5f3a1a62619cf3d/src/core/mac/sub_mac.cpp#L476-L480
I would propose to update the comment in `otPlatRadioReceive`'s description:
https://github.com/openthread/openthread/blob/950c7c861246554e539daa6197530c405d7b8028/include/openthread/platform/radio.h#L452-L462

> 
> > 5. `otPlatRadioReceive` can return `OT_ERROR_INVALID_ARGS` and `OT_ERROR_FAILED`, but these are not listed among the possible return values of this function in <openthread/platform/radio.h>. As a general question: are _other_ return values possible, apart from those stated in the header? Are the callers designed to fail "gracefully" due to generic/unhandled return errors?
> 
> OpenThread is designed to follow the documented radio driver state machine. There are `assert()` and logs to check when `otPlatRadioReceive()` returns anything other than `OT_ERROR_NONE`.
> 

Got it.

> OpenThread is designed to follow the documented radio driver state machine. There is an `assert()` to check when `otPlatRadioTransmit()` returns anything other than `OT_ERROR_NONE`.
Got it.

> `OT_RADIO_CAPS_TRANSMIT_RETRIES` is for retransmissions in response to no ack received. The name/docs could probably be updated to reflect this.

Ok, so the driver is fine; I'll propose some changes to the docs. IMHO the name is not bad.

> 
> > 9. I think the following should be `static`: `findSrcMatchAvailEntry`, `findSrcMatchShortEntry`, `findSrcMatchExtEntry`, `addToSrcMatchShortIndirect`, `addToSrcMatchExtIndirect`, `removeFromSrcMatchShortIndirect`, `removeFromSrcMatchExtIndirect`, etc.
> 
> Agree.

I'll submit a PR.

> 
> > 10. `removeFromSrcMatchExtIndirect`/`removeFromSrcMatchShortIndirect` use `memset` to clear the uint16_t checksum. Why? It would be simpler just to set `table[entry].checksum = 0;`.
> >          [openthread/examples/platforms/efr32/radio.c](https://github.com/openthread/openthread/blob/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999/examples/platforms/efr32/radio.c#L713-L716)
> >            Lines 713 to 716
> >          in
> >          [bf794ca](/openthread/openthread/commit/bf794ca5f27599f4d2ed8bfe71b9bb3cdd833999)
> >              
> >               void removeFromSrcMatchShortIndirect(uint8_t entry) 
> >               { 
> >                   srcMatchShortEntry[entry].allocated = false; 
> >                   memset(&srcMatchShortEntry[entry].checksum, 0, sizeof(uint16_t));
[... 58 lines cut ...]
> >               #if OPENTHREAD_ENABLE_DIAG 
> >                       if (otPlatDiagModeGet()) 
> >                       { 
> >                           otPlatDiagRadioReceiveDone(aInstance, &sReceiveFrame, sReceiveError); 
> >                       } 
> >                       else 
> >               #endif 
> >                       { 
> >                           // signal MAC layer for each received frame if promiscous is enabled 
[... cut ...]
> >                       } 
> >    
> >    `processNextRxPacket` first discards any packet with `length != IEEE802154_ACK_LENGTH` (actually, this block is only hit if `length >= IEEE802154_MIN_LENGTH && length <= IEEE802154_MAX_LENGTH`. So I think line 902 can be simplified to `if (sPromiscuous)`.
> 
> Agree.

Ok, I'll submit a PR once I get to test the modification. Actually, I meant that `sReceiveFrame.mLength > IEEE802154_ACK_LENGTH` is always true, so we can remove the `if(...)` altogether.

Thanks!
Diego.

jwhui, 2019-05-23:
@dismirlian, have you made progress on the items you listed above?

dismirlian, 2019-05-24:
Hi @jhwhui, I'm still waiting for the hardware to test. If you want, you can close this issue... all of my question have been answered, and I'll submit a PR in some weeks.

nigelg-silabs, 2019-06-06:
Hi @dismirlian, Should this ticket be closed? It seems to have been re-opened

jwhui, 2019-06-07:
@dismirlian, thanks for the update. Given that all of the questions have been answered, I'll close this issue. Feel free to submit new issues in the future.

closed 2019-06-07 by jwhui