openthread/openthread #3648: [posix-app] Unify posix-app API prefix

bukepo, 2019-03-06:
**Is your feature request related to a problem? Please describe.**
platform specific API prefix.

**Describe the solution you'd like**
A specific prefix for platform provided APIs. This helps caller tell what APIs are designed to be public and whether they are platform agnostic. Take posix-app platform as example, add `otPosix` to APIs for main.c

**Describe alternatives you've considered**
None.

**Additional context**
None.

sravanti298, 2019-09-12:
Observed that in the following commit https://github.com/openthread/openthread/pull/4073 
otSysGetTime() API is replaced with platformGetTime() and 
otSimSendRadioSpinelWriteEvent API is replaced with platformSimSendRadioSpinelWriteEvent
Are you expecting prefix name as otPosix or platform, which would be more helpful?
Kindly share your valuable suggestions to work on this task. Thanks!

bukepo, 2019-09-17:
In commit #4073, platform's non-public functions was renamed by removing the `ot` prefix. And public APIs are prefixed with `otSys`. I think `otPosix` looks better. If we change the prefix, maybe we should also rename `openthread-system.h` to `openthread-posix.h`. Thoughts? @jwhui

jwhui, 2019-09-17:
`otPosix` works for me. We used `otSys` in `examples/` for the platform abstraction. However, `otPosix` is specific to the POSIX app.

sravanti298, 2019-09-17:
Thanks for the support.
Currently `openthread-system.h` is being used by the platforms such as `POSIX, efr32mg12 and efr32mg21`.
So, replacing `otSys` with `otPosix` will effect other platforms (**efr32mg12, efr32mg21**).
For platform specific API’s, Shall I use `platform` prefix for all the hardware platforms in `openthread-master\examples\platforms` directory?
Please provide your valuable suggestions.

sravanti298, 2019-09-19:
@jwhui @bukepo Thanks for the support.
Iam just summarizing the requirements of this issue.
1. For platform specific API's remove `ot` prefix  
     - In the current openthread code-base,platforms such as `KW41Z`,`POSIX` and others are using the prefix names as  `kw41zUartProcess()`, `platformUartProcess()`,`efr32AlarmProcess()` and soon.
     - Since there is no `ot` prefix used, this requirement is already satisfied.
2. For public API's use `otPosix` prefix
     - Currently `openthread-system.h` is being used by the platforms such as `POSIX`, `efr32mg12` and `efr32mg21`.
     - So, renaming `openthread-system.h` to `openthread-posix.h` and `otSys` to `otPosix` will affect other platforms.

Please provide your valuable suggestions to proceed further.

jwhui, 2019-09-20:
@sravanti298, this issue applies to the posix-app source in [`src/posix`](https://github.com/openthread/openthread/tree/master/src/posix), not the code in [`examples/`](https://github.com/openthread/openthread/tree/master/examples/).

jwhui, 2020-02-20:
@bukepo, given that the `src/posix/platform` files have transitioned to C++, should we use C++ mechanisms for name spacing?

jwhui, 2023-06-27:
Closing stale issue.

closed 2023-06-27 by jwhui