openthread/openthread #3650: RFC7252 features (CoAP Service Discovery, Multicast All CoAP Nodes)

tarakanov, 2019-03-06:
I cannot find answer anywhere else, so put it here. 
I want to use RFC7252 features - service (Seciton 7) and multicast for "All CoAP nodes" (FF0x::FD, Section 12.8), to imlement M2M scenario. As I look through the OT code, I cannot find this implementation. 
Are this coap-features implemented? Maybe OT has alternative for that purpose (to multicast all OT or CoAP nodes and discover their resources)?

jwhui, 2019-03-06:
OpenThread does not, by default, subscribe to the All CoAP Nodes address. Take a look at the [otIp6SubscribeMulticastAddress()](https://openthread.io/reference/group/api-ip6#otip6subscribemulticastaddress) API for subscribing to arbitrary IPv6 multicast addresses.

tarakanov, 2019-03-07:
Ok, otIp6SubscribeMulticastAddress()  is more general approach, I will try it. 
And what about CoAP Service Discovery feature? Is it out of scope of current OT? Maybe OT has similar mechanism to discover node resources?
And one more great CoAP feature is observe mechainsm (rfc7641). Does OT has something similar?

jwhui, 2019-03-07:
OpenThread does not currently implement CoAP Observe or Service Discovery. Possible options to implement them are:
1. Implement the feature directly in OpenThread core,
1. Implement the feature directly in your application using the [otCoapSetDefaultHandler()](https://openthread.io/reference/group/api-coap#otcoapsetdefaulthandler) API,
1. Use a third-party CoAP library on top of OpenThread.

tarakanov, 2019-03-07:
Got it. Thank you

closed 2019-03-07 by tarakanov