openthread/openthread #3651: [spinel] add service spinel bindings

josephn123, 2019-03-06:
I will begin working on this. Please assign to me.

Firstly, a couple of questions:

- The most suitable Spinel property? `SPINEL_PROP_THREAD_SERVICE = SPINEL_PROP_THREAD__BEGIN + 15`?
- Should adding/removing a property over spinel trigger `otServerRegister()` or should this be done separately (like the CLI)?
- We want the ability to add, remove, list all services & list all leader services?

josephn123, 2019-03-06:
With reference to the above, I have decided to go with adding a new group of spinel properties

```
    SPINEL_PROP_SERVER__BEGIN = 0x90,

    SPINEL_PROP_SERVER_SERVICES = SPINEL_PROP_SERVER__BEGIN + 0,

    SPINEL_PROP_SERVER_ALLOW_LOCAL_DATA_CHANGE = SPINEL_PROP_SERVER__BEGIN + 1,

    SPINEL_PROP_SERVER__END = 0xA0,
```

Nopik, 2019-03-07:
@josephn123 can you outline a bit how you're going to implement this, please?

Generally it should be possible to register a number of services, providing:
a) enterprise number
b) service data (sequence of bytes)
c) server data (sequence of bytes)
d) stable flag
for each service.

Theoretically it should be possible to register a number of servers (multiple server data) for single service, though such flexibility will probably not be necessary in most use cases.

I'm not sure how the spinel props you listed above map to those data.

Nopik, 2019-03-07:
And - it should be possible to read:
a) whether service is successfully registered or not (i.e. confirmed by leader)
b) what aloc was assigned

The above can be read from network data, though it would be nice to not force user to parse it manually.

josephn123, 2019-03-08:
@nopik I've edited the above comment with updated spinel props and documentation.

Essentially these just call the `otServer*` APIs listed in `server_api.cpp`

jwhui, 2019-05-02:
Resolved by #3659 and https://github.com/openthread/wpantund/pull/386.

closed 2019-05-02 by jwhui