openthread/openthread #3656: [nrf52840] otPlatRadioDisable() doesn't disable radio

bukepo, 2019-03-08:
**Describe the bug**
From code, disabling nrf52840 doesn't really change the radio state in nordic SDK. If the radio state is in RECEIVE mode, otPlatRadioDisable() may still ACK packets.

LuDuda, 2019-03-08:
Thanks, @bukepo for reporting this.

According to the [documentation](https://github.com/openthread/openthread/blob/master/include/openthread/platform/radio.h#L215), OT should always call `otPlatRadioSleep()` before invoking `otPlatRadioDisable()`. For complete radio shutdown you can use `nrf5RadioDeinit()` function.

Do you think this behaviour should be modified (as well as OpenThread documentation allows for such transition)? I took a quick look on the first reference hardware platform, which seems to act in the same way.

bukepo, 2019-03-08:
Yes. Shall we add asserts to prevent illegal usage?

LuDuda, 2019-03-08:
Can you clarify your reply (I see two contradicted sentences :smile:)?

My understanding is that `otPlatRadioDisable` behaves correctly (according to the documentation). Do you think about adding assert in the platform code or OT?

bukepo, 2019-03-09:
Yes, I think we should add assert in platform code to prevent we use it in wrong way.

LuDuda, 2019-03-09:
Thanks, I will look at it and propose PR.

PR #3756 (2019-04-15): [platform] return error on incorrect otPlatRadioDisable usage
This PR fixes #3656.

Additionally, a small inefficiency with the nRF5x platform using FreeRTOS and multiprotocol (Thread + BLE) has been fixed.
closed 2019-04-15 by jwhui