openthread/openthread #3668: [udp] shall we return invalid arguments on port mismatch?

bukepo, 2019-03-12:
**Is your feature request related to a problem? Please describe.**
Source port is set to the socket's bound port directly without checking the desired port in aMessageInfo in https://github.com/openthread/openthread/blob/master/src/core/net/udp6.cpp#L186.

**Describe the solution you'd like**
Return OT_ERROR_INVALID_ARGS if aMessageInfo.mSockPort is non-zero and different from the socket's bound port.

**Describe alternatives you've considered**
use assert.

**Additional context**
To prevent invalid usages.

jwhui, 2019-03-12:
I believe the current OpenThread behavior is consistent with [RFC 3542 Section 6.2](https://tools.ietf.org/html/rfc3542#section-6.2):
```
   When specifying the source IPv6 address as ancillary data, if the
   ipi6_addr member of the in6_pktinfo structure is the unspecified
   address (IN6ADDR_ANY_INIT or in6addr_any), then (a) if an address is
   currently bound to the socket, it is used as the source address, or
   (b) if no address is currently bound to the socket, the kernel will
   choose the source address.  If the ipi6_addr member is not the
   unspecified address, but the socket has already bound a source
   address, then the ipi6_addr value overrides the already-bound source
   address for this output operation only.
```

bukepo, 2019-03-12:
Thanks for sharing this. I'm not sure if I understood the line right.
> then the ipi6_addr value overrides the already-bound source
   address for this output operation only

Our implementation seems did the reverse override. Does it means we should only call the following when messageInfoLocal is `0`?
```cpp
messageInfoLocal.SetSockPort(GetSockName().mPort);
```

jwhui, 2019-03-12:
Oh yes, somehow I thought this was about IPv6 Source Address rather than UDP Source Port.

RFC 3542 doesn't cover UDP ports and I'm not aware of any existing precedent. I'm fine with returning error if the socket is bound, the source port is non-zero, and the socket name port differs.

PR #3698 (2019-03-20): [udp] return error on source port mismatch in Udp::SendTo()
Resolves #3668.
closed 2019-03-20 by jwhui