openthread/openthread #3676: [code-style] unify shell scripts style

bukepo, 2019-03-12:
**Is your feature request related to a problem? Please describe.**
Styles of shell scripts differs.

**Describe the solution you'd like**
Use [google's shell style guide](https://google.github.io/styleguide/shell.xml).

**Describe alternatives you've considered**
None.

**Additional context**
I found the style guide is very helpful to write robust and easy understanding scripts.

praneethpanasala, 2020-02-14:
Hi Yakun Xu,

Good Day!

I am interested to work on this issue.
Could you please let me know whether this issue is applicable to the below script files in openthread code-base or not?

   ``` ./etc/docker/environment/docker-entrypoint.sh
     ./.travis/after_success.sh
     ./.travis/script.sh
     ./.travis/before_install.sh
     ./script/clang-format.sh
     ./script/clang-format-check.sh
     ./tools/harness-automation/start.sh
     ./tools/gerrit/git-squash-merge.sh
     ./tests/toranj/build.sh
     ./tests/toranj/start.sh 
```

Thanks in advance,
Praneeth Panasala.

jwhui, 2026-09-01:
Resolved by #4966

closed 2026-09-01 by jwhui