openthread/openthread #3679: [platform-netif] allow setting interface name from command line

bukepo, 2019-03-12:
**Is your feature request related to a problem? Please describe.**
now the interface name of platform-netif is generated from kernel. sometimes we need give the interface a name.

**Describe the solution you'd like**
add a command line option to accept the desired interface name and exit with error if unable to use that name.

**Describe alternatives you've considered**
None.

**Additional context**
prepare for more complicated tests in ot-br-posix.

VijayGitCode, 2019-09-11:
Hi Bukepo,
Could you please give more information ? 

1. The interface you are talking about is the interface name in border router(OTBR)?
2. Currently the interface name in OTBR is "wpan0"
3. A command line option is expected as part of "wpanctl"?

Please give your comments for the above queries. It will be helpful to work on this issue.

jwhui, 2019-09-17:
@VijayGitCode, this issue is specific to the [POSIX App](https://github.com/openthread/openthread/tree/master/src/posix) implementation, where OpenThread runs on a Linux-based host. The tun interface is managed within [src/posix/platform/netif.cpp](https://github.com/openthread/openthread/blob/87ef3b1d5105ac6225c9043a7be0137cdcc11cd5/src/posix/platform/netif.cpp#L371).

This GitHub issue seeks to add a command-line option to POSIX App that allows for specifying the network interface name.

jwhui, 2020-02-20:
Resolved by #4573.

closed 2020-02-20 by jwhui