openthread/openthread #3681: [nrf52840] unable to use USB=1 ot-ncp-radio

bukepo, 2019-03-12:
**Describe the bug**
/dev/ttyACM0 not found with USB=1 ot-ncp-radio.

**To Reproduce**
Information to reproduce the behavior, including:
1. dd69fabd2d42f51a05f2f752d0ff07566c327c68
2. nRF52840-dk
3. make -f examples/Makefile-nrf52840 USB=1
4. None

**Expected behavior**
/dev/ttyACM0 present just like the ot-ncp-ftd mode

**Console/log output**
```
[  369.039514] usb 1-1.2: device descriptor read/8, error -110
[  374.239518] usb 1-1.2: device descriptor read/8, error -110
[  374.459503] usb 1-1.2: new full-speed USB device number 13 using dwc_otg
[  379.519572] usb 1-1.2: device descriptor read/8, error -110
[  384.719541] usb 1-1.2: device descriptor read/8, error -110
[  384.839574] usb 1-1-port2: unable to enumerate USB device
[  541.109420] usb 1-1.2: new full-speed USB device number 14 using dwc_otg
[  546.319447] usb 1-1.2: device descriptor read/all, error -110
[  546.419422] usb 1-1.2: new full-speed USB device number 15 using dwc_otg
[  551.539426] usb 1-1.2: device descriptor read/64, error -110
```

**Additional context**
CDC is connected to raspberry PI.
```
Raspbian GNU/Linux 9 \n \l
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
```

bukepo, 2019-03-12:
Any idea for this issue?

LuDuda, 2019-03-12:
@bukepo thank you for raising this issue.

We do have on hardware CI tests triggered on OT commit, where we perform smoke tests i.e. of RCP with USB, however on Linux PC instead of Raspberry Pi that pass. I've just verified that on my PC it also works as expected. We will, of course, check it on Rasberry Pi, however, the fact that normal NCP image works as expected is very strange (the USB port is exactly the same). Could you do a couple things in the meantime?

1. Check on your PC (instead of RaspberryPi) if USB device is correctly enumerated with the same image.
2. What is exact version of nRF52840-DK (e.g. 1.0.0) and of Raspberry Pi (e.g. 3B+)?
3. Increase value of [USB_HOST_UART_CONFIG_DELAY_MS](https://github.com/openthread/openthread/blob/master/examples/platforms/nrf52840/platform-config.h#L347) e.g. to 500ms to see if the problem is related to the slow host. Indeed the error -110 indicates the timeout error.
4. Add LINK_RAW=1 flag, so that RCP can work correctly (this is step 2).

bukepo, 2019-03-13:
@LuDuda I'm using nRF52840-DK(0.9.2) and Raspberry Pi(3B+). And I found USB=1 cannot work on latest master. The working commit is 84a2706393e083508d591492b734d1f589f29ecc. With this version, both `ot-ncp-radio` and `ot-ncp-ftd` works.

bukepo, 2019-03-13:
@LuDuda changing USB_HOST_UART_CONFIG_DELAY_MS doesn't help.

LuDuda, 2019-03-13:
@kasroka can you look at it?

KaSroka, 2019-03-13:
@LuDuda Sure.

KaSroka, 2019-03-13:
@bukepo Ok, so the issue is probably due to board revision - you are using 0.9.2 which have some USB issues. Can you try with 0.11.0 or newer?

bukepo, 2019-03-13:
No, I only have this version. As I mentioned, if I flash an older version, it works as expected. I think there is bug in the code. Besides, it cannot be enumerated on my macOS either. We have a many 0.9.2 version boards. Could you please provide a fix?

KaSroka, 2019-03-13:
@bukepo After playing with `git bisect` I've found out that commit 203bce7 broke support for board revision 0.9.2. On 0.11.0 it is working perfectly fine. This commit changed nrf_drv_usbd to nrfx_usbd - they should work in same way. I'll keep you informed when I found out something.

KaSroka, 2019-03-14:
All I know right now is that it affects all Engineering A chips (QIAAAA marking on chip).

bukepo, 2019-03-19:
Any update for this?

LuDuda, 2019-03-19:
@bukepo this issue is currently handled internally by PM. For a quick fix we recommend to revert the commit: `git revert 203bce7ae79ac63652c27aa26b3d480cd9c19cc0`

LuDuda, 2019-04-12:
@bukepo after discussing the topic with PM, nRF52840 Engineering A chip support is end of line. Nordic recommends to switch to nRF52840 rev. C or later (production chip). I will update the README.md of our platform to state this. I believe we can close the issue now.

jwhui, 2019-04-23:
README was updated in #3759.

closed 2019-04-23 by jwhui