openthread/openthread #3699: [nRF52811] hang in radio mode

bukepo, 2019-03-20:
**Describe the bug**
Either working with pyspinel as sniffer or posix-app as radio transceiver, it get stuck in several minutes.

**To Reproduce**
Information to reproduce the behavior, including:
1. OPENTHREAD/0.01.00; NRF52811; Mar  3 2019 14:55:49
2. nRF52811
3. Build with USB=0 and baud rate=460800 

**Expected behavior**
It should not stuck or hang.

**Console/log output**
None.

**Additional context**
I tried another problem, it doesn't hang.

LuDuda, 2019-03-22:
Hello @bukepo,

You wrote: 
> I tried another problem, it doesn't hang.

Do you mean a different baud rate? Do you use DK with JLink?

Can you check if you forced HWFC and disabled MSD?
https://github.com/openthread/openthread/tree/master/examples/platforms/nrf52811#hardware-flow-control-detection

bukepo, 2019-03-22:
@LuDuda sorry I had a typo. I meant another platform. I will try your suggestion, and will update you when I get some results.

LuDuda, 2019-03-22:
@bukepo thanks for the update. My sniffer on RCP has been working for more than one hour now (in raw mode - not pcap mode). Can you please use GDB and send me the output whenever hang happens. Maybe we can figure out the problem if the board is in assert state.

LuDuda, 2019-03-31:
@bukepo Any update?

LuDuda, 2019-05-16:
Ping :smile:

bukepo, 2019-05-20:
not seeing this issue with latest code:+1:

closed 2019-05-20 by bukepo