openthread/openthread #3711: About Radio caps

dismirlian, 2019-03-23:
Hi all,

The following code says that `OT_RADIO_CAPS_TRANSMIT_RETRIES` implies `OT_RADIO_CAPS_CSMA_BACKOFF`:

https://github.com/openthread/openthread/blob/42c46368f191aae9637782943e72c52c4b11824b/src/core/mac/sub_mac.hpp#L434-L441

Shouldn't `OT_RADIO_CAPS_TRANSMIT_RETRIES` also imply `OT_RADIO_CAPS_ACK_TIMEOUT`, given that `OT_RADIO_CAPS_TRANSMIT_RETRIES` means that the radio performs the whole CSMA=>TX=>wait-for-ACK=>Retry-on-failure mechanism automatically?

Thanks,
Diego.

jwhui, 2019-03-23:
@dismirlian, yes `OT_RADIO_CAPS_TRANSMIT_RETRIES` implies `OT_RADIO_CAPS_ACK_TIMEOUT`. However, `OT_RADIO_CAPS_ACK_TIMEOUT` does not imply `OT_RADIO_CAPS_TRANSMIT_RETRIES`.

At the same time, it is valid for a radio to support `OT_RADIO_CAPS_CSMA_BACKOFF` but not `OT_RADIO_CAPS_CSMA_BACKOFF`.

jwhui, 2019-03-28:
@dismirlian, do you have further questions on this?

dismirlian, 2019-03-28:
Hi @jwhui, no, it's clear, thanks!

closed 2019-03-28 by jwhui