openthread/openthread #3743: build and flash openThread on FRDM-KW41Z

aymanelakhal, 2019-04-08:
GitHub Issues are for bugs and feature requests. To make bugs and feature requests more easy to find and organize, we close issues that are deemed out of scope for GitHub Issues.

The openthread-users Google Group is the recommended place for users to discuss OpenThread and interact directly with the OpenThread community. https://groups.google.com/forum/#!forum/openthread-users

aymanelakhal, 2019-04-08:
Hello,

I want to use the OpenThread stack on the FRDM-KW41Z, I started by building and generating the binary files, thereafter I flashed the binary file on FRDM-KW41z through (PE micro) Test Tool (nxp), when I want to connect to the bus uart (ttyACM0) to control the communication of Thread nothing happens as expected,
I did not make any changes or configuration on the git file of openthread, I don't know if something needs to be added in the configuration files,

Regards,

jwhui, 2019-04-09:
@aymanelakhal, I just tested the FRDM-KW41Z's MSD Bootloader method and that worked.

1. Clone the latest OpenThread repo.

2. Build the kw41z example apps:
```bash
$ ./bootstrap
$ make -f examples/Makefile-kw41z
```

3. Convert elf to bin.
```bash
$ cd output/kw41z/bin
$ arm-none-eabi-objcopy -O binary ot-cli-ftd ot-cli-ftd.bin
```

4. Connect the FRDM-KW41Z via USB, the FRDM-KW41Z should then appear as a drive on your computer.

5. Drag-and-drop the `ot-cli-ftd.bin` file into the drive. After copying, the drive should automatically unmount then mount again.

6. Open a serial connection and you should be able to interact with the CLI - I use `screen`.
```bash
$ screen <path-to-device> 115200
```

I have not used the NXP Test Tool, so cannot provide much help there.

Can you try the MSD Bootloader method?

aymanelakhal, 2019-04-10:
Hello @jwhui ,

Thanks for your feedback, I tested the drag-and-drop method and it worked, i don't know what was the problem, maybe the flash method through Test Tool,

Regards,

closed 2019-04-11 by aymanelakhal