openthread/openthread #3754: [dtls] on-mesh commissioner collapses if multiple dtls handshake with joiners

hjian2017, 2019-04-11:
If you use a joiner to repeatedly handshake with the on-mesh commissioner, join the thread network. When such a process is executed more than 15 times, it will cause a memory problem with the on-mesh commissioner and collapse.

sravanti298, 2019-09-10:
Trying to reproduce this issue in POSIX platform but, didn't find any memory problem.
When the joiner is successfully completed the commissioning process, it's entry is removing after several seconds.
Could you please provide some other details like which platform you have used?

jwhui, 2019-09-17:
@sravanti298, I believe @hjian2017 was using the nRF52840 SoC. That said, I'm not sure anyone has reproduced this recently. I will close this issue. We can reopen if it comes up again.

closed 2019-09-17 by jwhui