openthread/openthread #3755: nrf52840-dongle's  ot-ncp fails with uninitialized:fault

emenezes, 2019-04-11:
Hi folks, I know that a related issue was closed a few months ago (https://github.com/openthread/openthread/issues/3087) but I have the same problem here. Since it ended up straying a bit from the original problem and issuer did not follow up on the resolution, I'm asking again.

> * Commit ID for NCP firmware.
> * Commit ID for wpantund.
> * Build command/configuration you used for NCP firmware.

```
menezes@prime:~/wpantund$ git show
commit 42f9b6e553ebb0ed652ea8212b53c652865da19a (HEAD -> master, origin/master, origin/HEAD)
Author: Yakun Xu <xyk@google.com>
Date:   Sat Mar 30 02:08:09 2019 +0800

menezes@prime:~/openthread$ git show
commit 19544490e37a88642f5e016388038a9ac9def174 (HEAD -> master, origin/master, origin/HEAD)
Author: Jonathan Hui <jonhui@google.com>
Date:   Thu Mar 28 22:29:58 2019 -0700

menezes@prime:~/openthread$ make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 UDP_PROXY=1 BORDER_AGENT=1 LINK_RAW=1 SERVICE=1 USB=1 BOOTLOADER=1
menezes@prime:~/openthread$ arm-none-eabi-objcopy output/nrf52840/bin/ot-ncp-ftd -O ihex output/nrf52840/bin/ot-ncp-ftd.hex
```

I flashed the hex file using nrfconnect. 

The really funny thing is that I can connect to the NCP using pyspinel and the command _state_ returns _leader_ as it should. I also got working the NCP using the posix socket (ot-cli OR wpantund + ot-ncp) with the ot-ncp-radio firmware flashed in the dongle.

dmesg also seems right when I remove and insert back the dongle:

```
[mar29 19:37] usb 3-3: USB disconnect, device number 19
[  +7,976417] usb 3-3: new full-speed USB device number 20 using xhci_hcd
[  +0,150903] usb 3-3: New USB device found, idVendor=1915, idProduct=cafe
[  +0,000008] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000004] usb 3-3: Product: nRF52840 OpenThread Device
[  +0,000005] usb 3-3: Manufacturer: Nordic Semiconductor
[  +0,000004] usb 3-3: SerialNumber: E20EFXXXXXXX
[  +0,002014] cdc_acm 3-3:1.1: ttyACM0: USB ACM device
```

My wpantund.conf:

```
Config:NCP:SocketPath "/dev/ttyACM0"
Config:NCP:DriverName spinel
Daemon:SyslogMask "all"
```

Output from wpantund after reseting the NCP using wpanctl:

```
menezes@prime:~/wpantund$ sudo wpantund
wpantund[28434]: Starting wpantund 0.08.00d (Mar 29 2019 15:25:58) . . .
wpantund[28434]:     SOURCE_VERSION = 0.07.01-286-g42f9b6e
wpantund[28434]:     BUILD_VERSION = 0.07.01-286-g42f9b6e
wpantund[28434]: Configuration file "/etc/wpantund.conf" read.
wpantund[28434]: set-config-param: "Daemon:SyslogMask" set succeded
wpantund[28434]: Stat (autolog): Tx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[28434]: Stat (autolog): Tx history is empty
wpantund[28434]: Stat (autolog):
wpantund[28434]: Stat (autolog): Rx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[28434]: Stat (autolog): Rx history is empty
wpantund[28434]: Stat (autolog):
wpantund[28434]: Stat (autolog): NCP state history is empty.
wpantund[28434]: Stat (autolog):
wpantund[28434]: Stat (autolog):
wpantund[28434]: Opening tun interface socket with name "wpan0"
wpantund[28434]: Getting DBus connection
wpantund[28434]: Registering DBus connection
wpantund[28434]: Requesting DBus connection name com.nestlabs.WPANTunnelDriver
wpantund[28434]: Ready. Using DBUS bus ":1.183"
wpantund[28434]: Running as root without dropping privileges!
[... 83 lines cut ...]
wpantund[28434]: [-NCP-]: Multicast IPv6 address [1] "ff32:40:fdde:ad00:beef::1"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [2] "ff02::2"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [3] "ff03::2"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [4] "ff02::1"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [5] "ff03::1"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [6] "ff03::fc"

wpantund[28434]: Resetting and trying again... (retry 3)
wpantund[28434]: [->NCP] (6) RESET
wpantund[28434]: [NCP->] (0) PROP_VALUE_IS(LAST_STATUS) [70]
wpantund[28434]: [-NCP-]: Last status (RESET_POWER_ON, 112)
wpantund[28434]: [-NCP-]: NCP was reset (RESET_POWER_ON, 112)
wpantund[28434]: [->NCP] (7) PROP_VALUE_GET(PROTOCOL_VERSION)
wpantund[28434]: [NCP->] (7) PROP_VALUE_IS(PROTOCOL_VERSION) [0403]
wpantund[28434]: [->NCP] (8) PROP_VALUE_GET(NET_STACK_UP)
wpantund[28434]: [NCP->] (8) PROP_VALUE_IS(NET_STACK_UP) [01]
wpantund[28434]: [-NCP-]: Stack is up

wpantund[28434]: State change: "uninitialized" -> "associating"
wpantund[28434]: SpinelNCPInstance-Protothreads.cpp:430: Requirement Failed (!ncp_state_is_joining(get_ncp_state()))
wpantund[28434]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[28434]: State change: "associating" -> "uninitialized:fault"
wpantund[28434]: DBus Sending Interface Added Signal for wpan0
^C
```

jwhui, 2019-04-11:
@emenezes, thanks for raising this issue. Unfortunately, I am not able to reproduce this issue.

@LuDuda, any thoughts?

emenezes, 2019-04-12:
Thank you for the quick reply @jwhui.

> @emenezes, thanks for raising this issue. Unfortunately, I am not able to reproduce this issue.

Just to clarify, you reproduced the steps and it worked for you?

I have 2 dongles, tried on both of them in ubuntu 18.04 and RPi3 raspebian. I even tried the OTBR precompiled image, the webserver indicates wpantund disconnected.

LuDuda, 2019-04-12:
Hello @emenezes 

As @jwhui I am unable to reproduce this issue.

The one idea that I have in mind is to remove persistent storage from the device. Because it is a dongle with the built-in bootloader, we need to either connect external JLink programmer or simply erase OpenThread settings using `wpanctl`. Because the problem appears on the initialization phase in `wpantund`, can you try run `wpantund` with the following parameter (to prevent autostart):
`sudo wpantund -o Daemon:AutoAssociateAfterReset "false"`.

Then please use `sudo wpanctl leave` and `sudo wpanctl dataset erase` to remove persistent storage. Then try again.

Also, can you please download nRF5 SDK for Thread and Zigbee v3.0.0 and check the precompiled hex: `examples\thread\ncp\usb\hex\nrf52840_xxaa_mbr_pca10059.hex`

If that does not help, it would be great if @abtink can elaborate on `wpantund` requirement that is failing. To be honest i was able once or twice to get this error, but after next retry, `wpantund `always correctly initializes NCP.

emenezes, 2019-04-12:
Thanks @LuDuda, although your suggestion did not work directly it was in the right path. 

The commands  `sudo wpanctl leave` and `sudo wpanctl dataset erase` did not work. So I did the following:

1. updated Thread and Zigbee sdk
2. flashed the cli ftd example
3. connect to serial port and did a `factoryreset`
4. flashed the ncp ftd back

After that wpantund worked fine and I was able to follow the codelabs for nrf52840 and connect a cli device to a ncp device.

Once again, thanks for the quick reply and the helpful insight.

LuDuda, 2019-04-13:
That is great @emenezes! Although it would be even better to reveal the real root cause :smile: 

One thing we know is that persistent storage region that includes OpenThread settings is not automatically erased (on PCA10059 dongles) because bootloader only writes to the places where the application resides. I will try to figure out a better way than manually use `factoryreset` command to erase it. I am wondering if you had other (non-OpenThread based) application flashed before, that may potentially corrupt this persistent storage region, or its just wpantund that doesn't handle some cases.

emenezes, 2019-04-15:
@LuDuda I did not flash any other application, but since I have 2 dongles I mixed them pretty well. I'm certain that the one I used as ncp was flashed before with a cli hex and configured to work. The configuration there must be from that. Apart from it, I did not flash anything else.

Thanks for your help.

LuDuda, 2019-04-18:
Hello, @emenezes I was able to reproduce the problem by digging a little more into logs.

**Steps to reproduce:**
- Flash the `CLI FTD` application
- Become a Leader of the network
- Execute `autostart enable` command (any chance you used it?)
- Flash the `NCP FTD` (without erasing Non-Volatile OpenThread Settings - e.g. by using nRF52840 Dongle and nRF Connect)
- Run `wpantund` - which throws the same errors you posted

Because autostart option is enabled, the wpantund daemon does not solely manage the NCP and thus it throws a fail on suspicious state change. I haven't tried on POSIX but I feel the same error should be reproducible.

emenezes, 2019-04-19:
Yes!!! I did use the `autostart enable` when I flashed the CLI. I guess you just mapped another root cause problem.

BTW you guys are doing amazing work! Thanks for that and FYI, I'm a PhD candidate using OT to establish a flying IoT network to aid the recovery of disaster areas.

LuDuda, 2019-04-19:
@emenezes Thanks for the confirmation. The PhD thesis seems to be fantastic! I would be happy to read it and analyze conclusions after it is done :smile:

@jwhui Do you think this problem should be fixed somehow in OT itself (although note that its `otInstanceInitSingle` which starts thread part automatically, so it's not easy to bind it with `NCP`), or `wpantund` should be able to check `autostart` option before it starts joining procedure, or OpenThread users just need to be careful about it?

Anyway, I guess we can close this issue for now, and if need be create a new one as a feature request.

abtink, 2019-04-19:
@emenezes , OT autostart feature is not typically enabled/used when using it in NCP mode with `wpantund`. `wpantund` can take care of starting the network.  

Somewhat related to this: We have `"Daemon:AutoAssociateAfterReset"` configuration wpantund property to control `wpantund`'s behavior regarding this (it determines whether the `wpantund` should start the network (associate) if NCP is commissioned (there is a saved network) after NCP reset). Note that it's set to `true` by default. If set to `false` wpantund will leave this to application layer code (wait for it to issue `resume` command to restart the Thread network). 

Good luck on your PhD thesis... 100% agree with @LuDuda , it sounds amazing! ... :)

jwhui, 2019-04-19:
@abtink, while autostart feature is not typically used with NCP, it is possible to get into this state. I agree with @LuDuda that we should do something so that others do not run into this in the future.

closed 2019-04-23 by jwhui

sjlongland, 2019-06-12:
A heads up… we just hit this exact issue on CC2538 because our device stores the OpenThread settings off-chip in SPI flash.

So the problem isn't specific to nRF52840, it will affect any platform where the settings are not erased by flashing new firmware.  Maybe the NCP firmware could delete the `autostart` setting automatically when flashed, that would avoid this problem?

jwhui, 2019-06-14:
@sjlongland, just submitted #3920.

sjlongland, 2019-06-14:
@jwhui Fantastic. :-)  I'll have to give the new `wpantund` a try soon as our CI server picks up the changes.

PR #3920 (2019-06-14): [auto-start] remove auto-start feature
Resolves #3685.
Resolves #3755.