openthread/openthread #3768: assert triggered in otPlatRadioTransmit

NickNamerr, 2019-04-17:
**Describe the bug**
Firmware binary exited  [here](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/radio.c#L534)

After few hours of heavy load communications.

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id 2057a141ef75247197634d709bff4789d34c02d3
2. IEEE 802.15.4 hardware platform EFR32MG13p732GM48
3. Build steps : https://github.com/openthread/openthread/tree/master/src/posix
4. Network topology : 15 FTD and 20 MTD

**Expected behavior**
Return error and try to repeat transmission.

**Console/log output**
Does not have.

**Additional context**
UART running on 1.250.000 Baudrate

jwhui, 2019-04-17:
@NickNamerr, thanks for reporting this issue.

Just to confirm, the build steps you referenced indicate that you are using OpenThread POSIX app and radio-only firmware on the NCP, is that correct?

NickNamerr, 2019-04-17:
@jwhui Yes, this is correct.

jwhui, 2019-04-17:
@josephn123 @marvensilabs, any thoughts on what might trigger [this assert](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/radio.c#L534) in `otPlatRadioTransmit()`?

NickNamerr, 2019-04-17:
I change code to :
    if (status != RAIL_STATUS_NO_ERROR){
		otPlatReset(aInstance);
	}
    //assert(status == RAIL_STATUS_NO_ERROR);
And setup Breakpoint in oZone to catch variables value. Will update here once it will happen again.

marvensilabs, 2019-05-09:
In the RAIL library we check a status register to see if it is transmitting and if so the Tx functions return RAIL_STATUS_INVALID_STATE.  It looks like it could be RAILCb_Generic is setting sTransmitBusy to false when perhaps it shouldn't.  The RAIL_EVENTS_TX_COMPLETION mask in the SDK does not include RAIL_EVENT_SCHEDULER_STATUS or RAIL_EVENT_RX_ACK_TIMEOUT which could be the cause of sTransmitBusy being set to false whilst it is still in a transmit state.

I'm looking into this further.

jwhui, 2019-07-02:
Resolved by #3922.

closed 2019-07-02 by jwhui

NickNamerr, 2019-07-22:
Recently I pulled commit "db8c4c83399e5df7514d2ef8d4a57423e5f1b39a" and bug appeared in FTD, before was only at NCP(RCP).

Previously I used version : "a07751ec0d7b02394670803e40dd080833d94b36" and never happened on FTD.

bstraehl, 2019-07-29:
The assert in otPlatRadioTransmit still happens here with commit ec6057c52474485cecded83b4031f10c8c6b1904 on a standard NCP and with BRD4161A. It's quite easily to reproduce with high UDP traffic.

hemanth-silabs, 2019-07-29:
@bstraehl Is it possible for you to share the steps to reproduce this assert? Do you have any other information on where exactly the code asserts?

bstraehl, 2019-07-30:
@NickNamerr Yes, the assert **status == RAIL_STATUS_NO_ERROR** in otPlatRadioTransmit fails. And even the values 
status = RAIL_STATUS_INVALID_STATE; //(2)
aFrame->mInfo.mTxInfo.mCsmaCaEnabled = 1;
are the same as you mentioned. 

@hemanth-silabs For some reasons, which I never will understand, our company decided to implement another serial protocol for the NCP. It's spinel-like but it's not spinel. It's compatible with nothing and therefore it's not possible to run all the standard test cases and of course it doesn't work with wpantund. However I wrote my own test suite in Python3. I attached one of the test cases, which creates increasing UDP load between a leader and an end device. It tells you how I initialise the two devices and how the test is done. Maybe you can reproduce something similar with your own techniques. This test case fails randomly at different UDP payload sizes, because of the assert mentioned above.
[udp_test.zip](https://github.com/openthread/openthread/files/3445550/udp_test.zip)

NickNamerr, 2019-07-31:
@hemanth-silabs please check this [issue](https://github.com/openthread/openthread/issues/3768#issuecomment-483920532)

hemanth-silabs, 2019-08-05:
@NickNamerr @bstraehl Thank you for all the inputs. I think one of the changes to radio driver we made, did not make it to the PR. We are currently in process of raising another PR to fix this issue. I will let you know when the PR is raised.

hemanth-silabs, 2019-08-07:
@NickNamerr @bstraehl commit https://github.com/openthread/openthread/commit/6ebc5174087fbbe507f8132935d4f55ed7f6d1be fixes this issue.

NickNamerr, 2019-08-15:
@hemanth-silabs thanks!