openthread/openthread #3775: Multicast to Realm local address

vervaekejonathan, 2019-04-19:
Hi! 
Using OTBR in combination with nrf52 thread nodes.

I can ping to link local multicast addresses, but not to realm local multicast address.
Why is this and how can I discover all connected nodes in my thread network?
Thanks!

> pi@raspberrypi_otbr:~ $ **sudo wpanctl getprop "IPv6:MulticastAddresses"**
> IPv6:MulticastAddresses = [
> 	"ff02::1                                  origin:ncp"
> 	"ff02::2                                  origin:ncp"
> 	"ff03::1                                  origin:ncp"
> 	"ff03::2                                  origin:ncp"
> 	"ff03::fc                                 origin:ncp"
> 	"ff32:40:fdde:ad00:beef::1                origin:ncp"
> 	"ff33:40:fdde:ad00:beef::1                origin:ncp"
> ]

> pi@raspberrypi_otbr:~ $ **ping6 -c3 ff02::2%wpan0** 
> PING ff02::2%wpan0(ff02::2%wpan0) 56 data bytes
> 64 bytes from fe80::48dd:860a:9bed:9da7%wpan0: icmp_seq=1 ttl=64 time=0.170 ms
> 64 bytes from fe80::6d63:5136:8e40:e9c0%wpan0: icmp_seq=1 ttl=64 time=47.8 ms (DUP!)
> 64 bytes from fe80::a498:d8fe:925:7b3a%wpan0: icmp_seq=1 ttl=64 time=51.7 ms (DUP!)
> 64 bytes from fe80::2022:99d4:8eb2:a6f8%wpan0: icmp_seq=1 ttl=64 time=64.5 ms (DUP!)
> 64 bytes from fe80::48dd:860a:9bed:9da7%wpan0: icmp_seq=2 ttl=64 time=0.169 ms
> 64 bytes from fe80::2022:99d4:8eb2:a6f8%wpan0: icmp_seq=2 ttl=64 time=54.9 ms (DUP!)
> 64 bytes from fe80::a498:d8fe:925:7b3a%wpan0: icmp_seq=2 ttl=64 time=60.1 ms (DUP!)
> 64 bytes from fe80::6d63:5136:8e40:e9c0%wpan0: icmp_seq=2 ttl=64 time=65.3 ms (DUP!)
> 64 bytes from fe80::48dd:860a:9bed:9da7%wpan0: icmp_seq=3 ttl=64 time=0.147 ms
--- ff02::2%wpan0 ping statistics ---
3 packets transmitted, 3 received, +6 duplicates, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 0.147/38.344/65.371/27.510 ms

> pi@raspberrypi_otbr:~ $ **ping6 -c3 ff03::2%wpan0** 
> ping: ff03::2%wpan0: Name or service not known

> pi@raspberrypi_otbr:~ $ **ping6 -c3 ff32:40:fdde:ad00:beef::1%wpan0** 
> PING ff32:40:fdde:ad00:beef::1%wpan0(ff32:40:fdde:ad00:beef::1%wpan0) 56 data bytes
> 64 bytes from fe80::48dd:860a:9bed:9da7%wpan0: icmp_seq=1 ttl=64 time=0.174 ms
> 64 bytes from fe80::2022:99d4:8eb2:a6f8%wpan0: icmp_seq=1 ttl=64 time=56.1 ms (DUP!)
> 64 bytes from fe80::a498:d8fe:925:7b3a%wpan0: icmp_seq=1 ttl=64 time=62.5 ms (DUP!)
> 64 bytes from fe80::48dd:860a:9bed:9da7%wpan0: icmp_seq=2 ttl=64 time=0.155 ms
> 64 bytes from fe80::a498:d8fe:925:7b3a%wpan0: icmp_seq=2 ttl=64 time=46.8 ms (DUP!)
> 64 bytes from fe80::2022:99d4:8eb2:a6f8%wpan0: icmp_seq=2 ttl=64 time=52.0 ms (DUP!)
> 64 bytes from fe80::6d63:5136:8e40:e9c0%wpan0: icmp_seq=2 ttl=64 time=64.7 ms (DUP!)
> 64 bytes from fe80::48dd:860a:9bed:9da7%wpan0: icmp_seq=3 ttl=64 time=0.154 ms
> 
> --- ff32:40:fdde:ad00:beef::1%wpan0 ping statistics ---
> 3 packets transmitted, 3 received, +5 duplicates, 0% packet loss, time 2003ms
> rtt min/avg/max/mdev = 0.154/35.354/64.774/27.760 ms
> pi@raspberrypi_otbr:~ $ ping6 -c3 ff33:40:fdde:ad00:beef::1%wpan0 
> ping: ff33:40:fdde:ad00:beef::1%wpan0: Name or service not known

jwhui, 2019-04-23:
@vervaekejonathan, can you try specifying the interface using the `-I` argument instead? For example:
```
$ ping6 -Iwpan0 ff03::2
```

vervaekejonathan, 2019-04-24:
that rapports only the IP address of the OTBR:

> pi@raspberrypi_otbr:~ $ **ping6 -Iwpan0 ff03::2**
> PING ff03::2(ff03::2) from fd11:22:db8:2:48dd:860a:9bed:9da7 wpan0: 56 data bytes
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=1 ttl=64 time=0.169 ms
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=2 ttl=64 time=0.167 ms
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=3 ttl=64 time=0.177 ms
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=4 ttl=64 time=0.177 ms
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=5 ttl=64 time=0.158 ms
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=6 ttl=64 time=0.175 ms
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=7 ttl=64 time=0.145 ms
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=8 ttl=64 time=0.168 ms
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=9 ttl=64 time=0.096 ms
> 64 bytes from fd11:22:db8:2:48dd:860a:9bed:9da7: icmp_seq=10 ttl=64 time=0.126 ms
> ^C
> --- ff03::2 ping statistics ---
> 10 packets transmitted, 10 received, 0% packet loss, time 9374ms
> rtt min/avg/max/mdev = 0.096/0.155/0.177/0.029 ms

> pi@raspberrypi_otbr:~ $ **ping -Iwpan0 ff02::2**
> PING ff02::2(ff02::2) from fe80::48dd:860a:9bed:9da7%wpan0 wpan0: 56 data bytes
> 64 bytes from fe80::48dd:860a:9bed:9da7%wpan0: icmp_seq=1 ttl=64 time=0.183 ms
> 64 bytes from fe80::e0c3:872:6bcb:916c%wpan0: icmp_seq=1 ttl=64 time=46.2 ms (DUP!)
> 64 bytes from fe80::2022:99d4:8eb2:a6f8%wpan0: icmp_seq=1 ttl=64 time=51.4 ms (DUP!)
> 64 bytes from fe80::a498:d8fe:925:7b3a%wpan0: icmp_seq=1 ttl=64 time=57.9 ms (DUP!)
> 64 bytes from fe80::500:949c:7399:cdd8%wpan0: icmp_seq=1 ttl=64 time=65.9 ms (DUP!)
> 64 bytes from fe80::6d63:5136:8e40:e9c0%wpan0: icmp_seq=1 ttl=64 time=71.0 ms (DUP!)
> 64 bytes from fe80::48dd:860a:9bed:9da7%wpan0: icmp_seq=2 ttl=64 time=0.143 ms
> 64 bytes from fe80::500:949c:7399:cdd8%wpan0: icmp_seq=2 ttl=64 time=50.2 ms (DUP!)
> 64 bytes from fe80::e0c3:872:6bcb:916c%wpan0: icmp_seq=2 ttl=64 time=54.1 ms (DUP!)
> 64 bytes from fe80::a498:d8fe:925:7b3a%wpan0: icmp_seq=2 ttl=64 time=58.1 ms (DUP!)
> 64 bytes from fe80::2022:99d4:8eb2:a6f8%wpan0: icmp_seq=2 ttl=64 time=71.0 ms (DUP!)
> 64 bytes from fe80::6d63:5136:8e40:e9c0%wpan0: icmp_seq=2 ttl=64 time=84.9 ms (DUP!)
>

librasungirl, 2019-04-24:
Hi @vervaekejonathan 

In my trial on the setup (one OTBR Leader, one CLI Router), the ping could be forwarded into the Thread Network. And there are ping responses from both Leader and Router.

```
pi@raspberrypi:~/scripts $ ping -Iwpan0 ff03::2
PING ff03::2(ff03::2) from fd11:22::aa3e:9f49:9d9d:14ab wpan0: 56 data bytes
64 bytes from fd11:22::aa3e:9f49:9d9d:14ab: icmp_seq=1 ttl=64 time=0.188 ms
64 bytes from fd11:22::5462:5212:a6ed:66ef: icmp_seq=1 ttl=64 time=39.4 ms (DUP!)
64 bytes from fd11:22::aa3e:9f49:9d9d:14ab: icmp_seq=2 ttl=64 time=0.147 ms
64 bytes from fd11:22::5462:5212:a6ed:66ef: icmp_seq=2 ttl=64 time=16.7 ms (DUP!)
64 bytes from fd11:22::aa3e:9f49:9d9d:14ab: icmp_seq=3 ttl=64 time=0.139 ms
64 bytes from fd11:22::5462:5212:a6ed:66ef: icmp_seq=3 ttl=64 time=71.5 ms (DUP!)
^C
--- ff03::2 ping statistics ---
3 packets transmitted, 3 received, +3 duplicates, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 0.139/21.369/71.598/26.522 ms
```

[image]

One Note is that the default **hoplimit (i.e. ttl)** for multicast is 1 by default, so the ping, though destined to realm local address, is only forwarded twice by OTBR Leader. To properly propagate across the Thread Network, you may want to specify one proper hoplimit according to your topology, for example, `ping -Iwpan0 ff03::2 -t 5`

vervaekejonathan, 2019-04-24:
Hi @librasungirl 
Changing the hoplimit did not fix the problem :(
Thanks for the tip anyway, I'll use it once this issue is solved ;)

librasungirl, 2019-04-24:
@vervaekejonathan,  would you please update the log level of wpantund and check whether the ping request is passed to the NCP radio like below sample log? 

// fyi, update the log level
1) add one line `Daemon:SyslogMask "all -debug"` to `/etc/wpantund.conf`
2) then restart wpantund service `sudo systemctl restart wpantund`
3) check the syslog `tail -f /var/log/syslog | grep wpantund`

// sample log
> Apr 24 10:25:40 raspberrypi wpantund[4209]: wpantund[4209]: [->NCP] IPv6 len:104 type:58 [SECURE]
> Apr 24 10:25:40 raspberrypi wpantund[4209]: wpantund[4209]: #011to(remote):[ff03::2]
> Apr 24 10:25:40 raspberrypi wpantund[4209]: wpantund[4209]: #011from(local):[fd00:7d03:7d03:7d03:9897:d494:9b3c:88e7]

vervaekejonathan, 2019-04-24:
it seems restarting wpantund solved the issue.. 
thanks!

librasungirl, 2019-04-24:
Great to know it works finally. Thanks for the update.

closed 2019-04-24 by librasungirl