openthread/openthread #3782: Masterkey rotation

vervaekejonathan, 2019-04-24:
I would like to know more about masterkey rotation.

I can change the masterkey on the OTBR, but of course this key is not pushed to the nodes. 
Is there a way this can be pushed via wpanctl? Or does this require an own implementation?

librasungirl, 2019-04-24:
@vervaekejonathan , for one existing Thread Network, the valid operation to update masterkey is from commissioner via pending dataset which will take effect after specified delay (default min delay time 5min) so that there is chance for the dataset to propagate to Thread Network.

There are wpanctl commands which could help to promote OTBR to become on-mesh commissioner and update the pending dataset. 

Fyi, here is the sample usage.

> 
> sudo wpanctl -I wpan0 commissioner start
> // wait some time for commissioner petition process
> 
> sudo wpanctl -I wpan0 dataset erase
> sudo wpanctl -I wpan0 setprop Dataset:PendingTimestamp 000000000000000B
> sudo wpanctl -I wpan0 setprop Dataset:ActiveTimestamp 000000000000000A
> sudo wpanctl -I wpan0 setprop Dataset:MasterKey 00112233445566778899aabbccddffee
> sudo wpanctl -I wpan0 setprop Dataset:Delay 350
> sudo wpanctl -I wpan0 dataset mgmt-set-pending 
>

vervaekejonathan, 2019-04-24:
@librasungirl this works, thanks!

How safe is this? I mean, if someone has my current masterkey and is sniffing the data all the thread data, can the new masterkey be detected in the packets?

jwhui, 2019-04-24:
@vervaekejonathan, Thread 1.1.1 does not provide a mechanism to securely communicate key updates to specific Thread devices. If someone has the Thread master key, they can also decode key update messages.

closed 2019-04-26 by jwhui