openthread/openthread #3790: DHCPv6 support in OpenThread

kwangeom, 2019-04-29:
**Is your feature request related to a problem? Please describe.**
We want to use "dnsmasq" in BBB as dhcpv6 server instead of one running in NCP.
Current "dhcp6_clinet.cpp/.hpp" looks not support this case.
We have 3 problems with current "dhcp6_client.cpp/.hpp".
1. it use RLOC address of NCP as target address of Solicit message. So the message is always terminated at NCP, not delivered to BBB.
2. We changed the target address of Solicit message to multicast address, but NCP delivers the Solicit message from other devices to BBB, but it does not deliver its own Solicit message to BBB.
We want dhcpv6 client in NCP also to get ip address from "dnsmasq".
3. "dnsmasq" responds with Reply message to the thread device, but Reply message is rejected by the thread device because its format is different from one generated by "dhcp6_server.cpp".
Exactly speaking, ProcessServerIdentifier(aMessage, optionOffset) and ProcessIaNa(aMessage, optionOffset) are throwing error to this Replay message. So the thread device cannot get ip address allocation eventually.     

**Describe the solution you'd like**
1. please add multicast option for Solicit message for those who want to use any DHCPv6 server in BBB.
2. please let NCP deliver its own Solicit to BBB when multicast address is used.
3. please let thread device accept Reply from dnsmasq.

This is Reply from dnsmasq 👍 
[image]

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kwangeom, 2019-05-06:
do you have any plan on this ticket? I would like to support this feature ASAP. If you don't have plan in near future on this ticket, could you please give us patches to fix this issue?

Regards,

Kwang

jwhui, 2019-05-16:
@kwangeom, thank you for raising this issue.

Communicating using RLOCs between the NCP and host is a non-trivial task since they both need to share the same IPv6 address. Communicating IPv6 packets between the NCP and host is not something easily addressed in this architecture.

Have you considered using an architecture where OpenThread core is running on the host (see [OpenThread POSIX app](https://github.com/openthread/openthread/tree/1284384bd8141e3a5ff4d3dda4fef6cc7003f311/src/posix))? The POSIX app includes a `PLATFORM_NETIF` build switch that allows OpenThread to control the host network interface.

jwhui, 2019-07-01:
Closing stale issue.

closed 2019-07-01 by jwhui