openthread/openthread #3797: [efr32] wpantund Stopped with last UART and RADIO driver updates.

NickNamerr, 2019-05-06:
Wpantund keeps stopping with "[-NCP-]: Socket error on read: Input/output error 5"

To Reproduce
Information to reproduce the behavior, including:

Git commit id 267a3b3 
IEEE 802.15.4 hardware platform EFR32MG13p732GM48
Build steps : https://github.com/openthread/openthread/tree/master/src/posix
Network topology : 10 FTD 
Expected behavior
"wpantund" should not be stopped.

Console/log output
May 06 17:01:49  wpantund[19]: Starting wpantund 0.08.00d (Feb  4 2019 06:53:05) . . .
May 06 17:01:26  wpantund[19]:         SOURCE_VERSION = 0.07.01-274-g8e6a5cf
May 06 17:01:26  wpantund[19]:         BUILD_VERSION = 0.07.01-274-g8e6a5cf
May 06 17:01:26  wpantund[19]: Configuration file "/etc/wpantund.conf" read.
May 06 17:01:26  wpantund[19]: Ready. Using DBUS bus ":1.2"
May 06 17:01:26  wpantund[19]: Running as root without dropping privileges!
May 06 17:01:26  wpantund[21]: About to exec "/opt/pyfarm/asix-driver/ot-ncp /dev/ttyF0"
May 06 17:01:26  wpantund[19]: [-NCP-]: NCP was reset (RESET_POWER_ON, 112)
May 06 17:01:26  wpantund[19]: Unexpected reset during NCP initialization.
May 06 17:01:26  wpantund[19]: State change: "uninitialized" -> "offline"
May 06 17:01:26  wpantund[19]: Resetting and trying again... (retry 1)
May 06 17:01:26  wpantund[19]: State change: "offline" -> "uninitialized"
May 06 17:01:26  wait NCP: No child processes
May 06 17:01:26  wpantund[19]: [-NCP-]: NCP was reset (RESET_POWER_ON, 112)
May 06 17:01:26  wpantund[19]: State change: "uninitialized" -> "offline"
May 06 17:01:26  wpantund[19]: NCP is running "OPENTHREAD/20170716-01537-g267a3b3e-dirty; POSIX-APP; May  2 2019 13:53:38"
May 06 17:01:26  wpantund[19]: Driver is running "0.08.00d (0.07.01-274-g8e6a5cf; Feb  4 2019 06:53:05)"
May 06 17:01:26  wpantund[19]: Network is not joinable
May 06 17:01:26  wpantund[19]: State change: "offline" -> "offline:commissioned"
May 06 17:01:26  wpantund[19]: Finished initializing NCP
May 06 17:01:26  wpantund[19]: AutoResume is enabled. Trying to resume.
May 06 17:01:26  wpantund[19]: NCP is commissioned. Resuming...
May 06 17:01:26  wpantund[19]: State change: "offline:commissioned" -> "associating"
May 06 17:01:26  wpantund[19]: State change: "associating" -> "associated"
May 06 17:01:26  wpantund[19]: Node type change: "unknown" -> "router"
May 06 17:01:29  Setting network parameters...
May 06 17:01:29  wpantund[19]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
May 06 17:01:29  wpantund[19]: SendCommand task encountered an error: 4 (0x00000004)
May 06 17:01:29  setprop failed with error 4. InvalidForCurrentState
May 06 17:01:29  wpantund[19]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
May 06 17:01:29  wpantund[19]: SendCommand task encountered an error: 4 (0x00000004)
May 06 17:01:29  setprop failed with error 4. InvalidForCurrentState
May 06 17:01:29  wpantund[19]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
May 06 17:01:29  wpantund[19]: SendCommand task encountered an error: 4 (0x00000004)
May 06 17:01:29  setprop failed with error 4. InvalidForCurrentState
May 06 17:01:29  form failed with error 10. Already
May 06 17:01:29  Forming WPAN "TestNetwork" as node type "router", channel:15
May 06 17:01:29  Network setup done.
May 06 17:01:29  Using prefix "fddd::"
May 06 17:01:29  wpantund[19]: Adding on-mesh prefix "fddd::/64" to NCP
May 06 17:01:29  Gateway configured.
May 06 17:01:29  Adding route prefix "fdde::" with len 64, priority "medium", stable:yes, domain-id 0.
May 06 17:01:29  wpantund[19]: Adding off-mesh route "fdde::/64" with preference medium to NCP
May 06 17:01:29  Route prefix added.
May 06 17:01:29  wpantund
May 06 17:01:42  wait NCP: No child processes
May 06 17:01:42  wpantund[19]: [-NCP-]: Socket error on read: Input/output error 5
May 06 17:01:42  wpantund[19]: NCPInstance: errno 5 "Input/output error"
May 06 17:01:42  wpantund[19]: Cleaning up. (gRet = 9)
May 06 17:01:42  wpantund[19]: Stopped.

Additional context
UART running on 1.250.000 Baudrate, RAIL v2.5, kReceiveFifoSize = 4096 in uart.c

NickNamerr, 2019-05-06:
"wpanctl get" forcing "wpantund" to stop execution and output is:
May 06 17:17:29  wpantund[19]: Signal jamming cleared
May 06 17:17:29  wpantund[19]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
May 06 17:17:29  wpantund[19]: SendCommand task encountered an error: 16 (0x00000010)
May 06 17:17:33  wpantund[19]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
May 06 17:17:33  wpantund[19]: SendCommand task encountered an error: 15351836 (0x00EA401C)
May 06 17:17:33  wpantund[19]: [-NCP-]: RCP is running "OPENTHREAD/20170716-01537-g267a3b3e-dirty; EFR32; May  2 2019 16:41:35"
May 06 17:17:33  wpantund[19]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
May 06 17:17:33  wpantund[19]: SendCommand task encountered an error: 15335444 (0x00EA0014)
May 06 17:17:34  wait NCP: No child processes
May 06 17:17:34  wpantund[19]: [-NCP-]: Socket error on read: Input/output error 5
May 06 17:17:34  wpantund[19]: NCPInstance: errno 5 "Input/output error"
May 06 17:17:34  wpantund[19]: Cleaning up. (gRet = 9)
May 06 17:17:34  wpantund[19]: Stopped.

josephn123, 2019-05-06:
Hi @NickNamerr 

- Does everything work as expected if kReceiveFifoSize is the default of 128?
- If so, you need to ensure kDmaBlockSize is half of the kReceiveFifoSize, i.e. if kReceiveFifoSize = 4096, kDmaBlockSize = 2048. This isn't at all obvious from the code, so some compile time checks, or at least some comments should be added.

josephn123, 2019-05-06:
Hi @NickNamerr 

Does everything work as expected with a default kReceiveFifo size of 128?

NickNamerr, 2019-05-07:
Hi @josephn123 it does not, happened even often. My test it to send CoAP msg with payload 1037 bytes.

NickNamerr, 2019-05-07:
few more symptoms:
1) "Node type change: "unknown" -> "end-device" - but "RLOC" in routing parameters(OffMeshRoutes and OnMeshPrefix) did not update;
2) Caught assert at "RAILInt_xxxxxxxxx..." in oZone debugger';

josephn123, 2019-05-07:
Does this look related to your other bug report #3768 ? I'll test myself today

NickNamerr, 2019-05-08:
It does not relate directly, but behavior sometimes looks similar, RAIL API returns  busy state. 
How to enable logs for "POSIX <-> REAL Hardware " model?

NickNamerr, 2019-05-09:
@josephn123 "while true; do wpanctl get; sleep 2; done" - causes ot-ncp-radio firmware stuck in loop of method "RAILINT_xxxxxxxx". oZone debugger in disassembly windows show that there check R0 for 1 and it is 4, looks like SiLabs RAIL waits for something that never return right value.

jwhui, 2019-07-02:
Resolved by #3922.

closed 2019-07-02 by jwhui