openthread/openthread #3798: Single entropy source sharing between OpenThread and application

KaSroka, 2019-05-06:
Currently, it's hard to develop an application that uses the same entropy source as OpenThread platform (especially with some kind of RTOS). Also in some places, we use entropy source directly instead of using mbedtls random generator which is not a good idea. Entropy source should be only used as a seed for mbedtls random generator.

I propose to create a single instance of mbedtls entropy context that would be managed by OpenThread and shared with the application. Also, we may create an abstraction for random that would keep a single instance of mbedtls random generator context and provide proper API to OT stack. This would allow us to remove `otPlatRandomGetTrue` and `otPlatRandomGet` from platform and replace it with single `otPlatEntropyGet`. It would be used and managed only by mbedtls entropy module. Also, this would clarify the usage of entropy and random functions as the current name may be misleading.

KaSroka, 2019-05-06:
To clarify this abstraction for random would use mbedtls_ctr_drbg_rand (which with this entropy source) inside. This approach would also allow us to use mbedtls built-in mutex support in RTOS applications.

jwhui, 2019-05-07:
@KaSroka, this sounds like a great enhancement. Would you like to submit a PR?

KaSroka, 2019-05-07:
@jwhui Yes, I'm working on it right now

closed 2019-05-21 by KaSroka