openthread/openthread #3802: DLTS commissioning from mobile app.

gidhin, 2019-05-07:
**Describe the bug**
A clear and concise description of what the bug is.
Need to commissioning from hybrid mobile app. I have install "npm i node-dtls-client" and i follow the instruction but didn't work.

code: 
var dtls = require('node-dtls-client');

    dtls.setLogLevel(dtls.logLevel.FINE);

    var client = dtls.connect(4915, '10.42.0.6', 'udp4', function () {
      client.send(new Buffer('foo\n'));
    });

    client.on('message', function (msg) {
      console.log(msg);
    });

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior**
A clear and concise description of what you expected to happen.

**Console/log output**
If applicable, add console/log output to help explain your problem.
error getting : 

Error: ./node_modules/node-dtls-client/build/dtls.js
Module not found: Error: Can't resolve 'dgram' in '/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/node-dtls-client/build'
resolve 'dgram' in '/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/node-dtls-client/build'
  Parsed request is a module
  using description file: /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/node-dtls-client/package.json (relative path: ./build)
    Field 'browser' doesn't contain a valid alias configuration
  after using description file: /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/node-dtls-client/package.json (relative path: ./build)
    resolve as module
      looking for modules in /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules
        using description file: /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/package.json (relative path: ./node_modules)
          Field 'browser' doesn't contain a valid alias configuration
        after using description file: /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/package.json (relative path: ./node_modules)
          using description file: /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram/package.json (relative path: .)
            no extension
              Field 'browser' doesn't contain a valid alias configuration
              /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram is not a file
            .ts
              Field 'browser' doesn't contain a valid alias configuration
              /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram.ts doesn't exist
            .js
              Field 'browser' doesn't contain a valid alias configuration
              /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram.js doesn't exist
            .json
              Field 'browser' doesn't contain a valid alias configuration
              /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram.json doesn't exist
            as directory
              existing directory
                use ./index.js from main in package.json
                  using description file: /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram/package.json (relative path: .)
                    Field 'browser' doesn't contain a valid alias configuration
                  after using description file: /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram/package.json (relative path: .)
                    using description file: /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram/package.json (relative path: ./index.js)
                      no extension
                        Field 'browser' doesn't contain a valid alias configuration
                        /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram/index.js doesn't exist
                      .ts
                        Field 'browser' doesn't contain a valid alias configuration
                        /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/dgram/index.js.ts doesn't exist
                      .js
[... 61 lines cut ...]
              /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/nodejs-dtls.json doesn't exist
            as directory
              /Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/nodejs-dtls doesn't exist
[/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/nodejs-dtls]
[/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/nodejs-dtls.ts]
[/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/nodejs-dtls.js]
[/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/nodejs-dtls.json]
[/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/nodejs-dtls]
 @ ./src/pages/scan-wifi/scan-wifi.ts 14:11-33
 @ ./src/app/app.module.ts
 @ ./src/app/main.ts
    at new BuildError (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/@ionic/app-scripts/dist/util/errors.js:16:28)
    at callback (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/@ionic/app-scripts/dist/webpack.js:121:28)
    at emitRecords.err (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/webpack/lib/Compiler.js:265:13)
    at Compiler.emitRecords (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/webpack/lib/Compiler.js:371:38)
    at emitAssets.err (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/webpack/lib/Compiler.js:258:10)
    at applyPluginsAsyncSeries1.err (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/webpack/lib/Compiler.js:364:12)
    at next (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/tapable/lib/Tapable.js:218:11)
    at Compiler.compiler.plugin (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/webpack/lib/performance/SizeLimitsPlugin.js:99:4)
    at Compiler.applyPluginsAsyncSeries1 (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/tapable/lib/Tapable.js:222:13)
    at Compiler.afterEmit (/Users/vetri/Vetri's Work Space/Projects/BLE_Stack/BLE/node_modules/webpack/lib/Compiler.js:361:9)
M

**Additional context**
Add any other context about the problem here.

bukepo, 2019-05-07:
@gidhin From https://www.npmjs.com/package/node-dtls-client, it seems this module doesn't support the cipher suite required by Thread commissioning. I think you should try https://github.com/particle-iot/node-mbed-dtls.

gidhin, 2019-05-07:
I have try this one but not working for me.

error : Module not found: Error: Can't resolve 'dgram'
  "node-dtls-client/build"

code is not working for me.

let option = {
type: "udp4",
address: "10.42.0.6",
port: 4915,
psk: { "psk_hint": "PSK" },
timeout: 1000, // in ms, optional, minimum 100, default 1000
ciphers: ["TLS_PSK_WITH_3DES_EDE_CBC_SHA",
"TLS_PSK_WITH_AES_128_CBC_SHA",
"TLS_PSK_WITH_AES_256_CBC_SHA",
"TLS_PSK_WITH_AES_128_CBC_SHA256",
"TLS_PSK_WITH_AES_256_CBC_SHA384",
"TLS_PSK_WITH_AES_128_GCM_SHA256",
"TLS_PSK_WITH_AES_256_GCM_SHA384",
"TLS_PSK_WITH_AES_128_CCM_8",
"TLS_PSK_WITH_AES_256_CCM_8"]
}

const dtls = require('node-dtls-client');

const socket = dtls.createSocket(option).on("connected", () => {/* start
sending data */
console.log("dtls testing");
});

i'm working on ionic application
Thanks and Regards,
Gidhin S

jwhui, 2019-05-23:
@gidhin, from your note above, it seems you are still using `node-dtls-client`, which does not support the ciphersuite `TLS_ECJPAKE_WITH_AES_128_CCM_8 ` that Thread utilizes.

As @bukepo noted above, you can try with [`node-mbed-dtls`](https://github.com/particle-iot/node-mbed-dtls).

gidhin, 2019-05-24:
Thanks for the update. Will check and let you know.

Thanks and Regards,
Gidhin S

gidhin, 2019-07-17:
Hi
Im getting this error  :
The DTLS handshake timed out
    at Socket.expectHandshake

This is my code

var dtls = require("node-dtls-client").dtls;
options = {
type: "udp6",
address: "10.42.0.1",
port: 49888,
psk: {
"psk_hint": "PSK"
},
timeout: 1000, // in ms, optional, minimum 100, default 1000
ciphers: ["TLS_PSK_WITH_3DES_EDE_CBC_SHA",
"TLS_PSK_WITH_AES_128_CBC_SHA",
"TLS_PSK_WITH_AES_256_CBC_SHA",
"TLS_PSK_WITH_AES_128_CBC_SHA256",
"TLS_PSK_WITH_AES_256_CBC_SHA384",
"TLS_PSK_WITH_AES_128_GCM_SHA256",
"TLS_PSK_WITH_AES_256_GCM_SHA384",
"TLS_PSK_WITH_AES_128_CCM_8",
"TLS_PSK_WITH_AES_256_CCM_8",
] // optional array of (D)TLS cipher suites, e.g.
["TLS_PSK_WITH_AES_128_CCM"]
}

const socket = dtls.createSocket(options, function (err, res) {
console.log("err", err);
console.log("res", res);
})
// create a socket and initialize the secure connection
// .createSocket(options /* DtlsOptions */ )
// subscribe events
.on("connected", e => {
/* start sending data */
console.log("connected");
console.log("CONNECTED", e);
})
.on("error", e => {
console.log("error gid", e);
})
.on("message", (msg /* Buffer */ ) => {
console.log("message", msg);
})
.on("close", e => {
console.log("close", e);
});

How can i pass message for handshaking ?

Thanks and Regards,
Gidhin S

bukepo, 2019-07-17:
Hi @gidhin It seems you are still using node-dtls-client instead of node-mbed-dtls.

jwhui, 2019-08-06:
Closing stale issue.

closed 2019-08-06 by jwhui