openthread/openthread #3806: ModuleNotFoundError: No module named 'Crypto'

sreeramp96, 2019-05-09:
I have installed Crypto using pip, but it still shows this error

[image]
 from Crypto.PublicKey import RSA
ModuleNotFoundError: No module named 'Crypto'

jwhui, 2019-05-09:
This question was also raised in #1137.

You need to install pycryptodome version 3.4.3:
```bash
$ pip install pycryptodome==3.4.3
```

You also need to uninstall `pycrypto` to avoid the name collision - both `pycrypto` and `pycryptodome` have a `Crypto` module.

sreeramp96, 2019-05-09:
okay will check this

closed 2019-05-13 by sreeramp96

MagnusAycox, 2022-04-11:
I have installed pycryptodomex and checked that crypto and pycrypto is uninstalled.
Still get "ModuleNotFoundError: No module named 'Crypto'"
I have Python version 3.9.2.

cberetta, 2023-03-20:
> I have installed pycryptodomex and checked that crypto and pycrypto is uninstalled. Still get "ModuleNotFoundError: No module named 'Crypto'" I have Python version 3.9.2.

Hi! I've found an explanation for this problem, it's all into the naming: "pycryptodome" and "pycryptodome**x**" _(note the final "x")_ are using different names: one use Crypto and the other use Cryptodome... 
Basically you have to use "from Crypto import ..." if you use pycryptodome or use "from Crypodome import ..." if you use pycryptodome**x**

Major information in this question I've found on Stackoverflow that explain the difference: https://stackoverflow.com/questions/48155294/what-is-pycryptodomex-and-how-does-it-differ-from-pycryptodome