openthread/openthread #3807: Bug: SED with multiple IP addresses fails to attach (continuously)

abtink, 2019-05-10:
Discovered this issue while analyzing some failed cases of `toranj` test case [`test-010-on-mesh-prefix-config-gateway`](https://github.com/openthread/openthread/blob/master/tests/toranj/test-010-on-mesh-prefix-config-gateway.py). 

An detached SED child with multiple IPv6 addresses can fail to attach back to its original parent (or any new router/parent).

- In MLE "Child ID Request" the child will include `Address Registration TLV` to register its IPv6 addresses with the parent.
- MLE messages do not use link-layer security, unless the message requires fragmentation at 6lowpan layer.
- The registration of IPv6 addresses can increase the "Child ID Request" message size, requiring fragmentation and therefore link-layer security.
- A parent node will reject secure frames from a new child trying to attach (note that link/MLE frame counters of the child are part of the "Child ID Request" message itself).

I am working on a solution/fix (along with a specific test-case to cover this situation). 
The issue seems to happen with 3 external addresses (in addition to mesh-local)

librasungirl, 2019-05-10:
@abtink there should have been some discussions within Thread Group about the issue you described.
And it should have agreed that if the AddressRegistraionTLV during attach (Child Id Request) or reattach (Child Update Request to restore link with parent) causing fragments, the additional addresses except MLEID could be postponed to register via Child Update Request after attach/reattach finishes.

abtink, 2019-05-10:
Ype. Thanks. we are going for same solution.

abtink, 2019-05-10:
Submitted https://github.com/openthread/openthread/pull/3813

jwhui, 2019-05-16:
Resolved by #3813.

closed 2019-05-16 by jwhui