openthread/openthread #3815: No response of <networkdiagnostic get> command when FTD CLI becomes leader?

tarunsmalviya, 2019-05-14:


jwhui, 2019-05-14:
@tarunsmalviya, thanks for raising this issue. However, I was not able to reproduce this behavior. In my simple setup of 2 FTDs, I am able to perform the following:
```
> state
leader
Done
> networkdiagnostic get fdde:ad00:beef:0:ad0a:f8d0:235e:b1a6 0 1 5 6 16
DIAG_GET.rsp/ans: 00088ee0896d0b4fad4201028400050bf60000800040000000f101060870ceb8ac40912e101000
```

Can you provide more details to help us reproduce?

jwhui, 2019-05-23:
@tarunsmalviya, can you provide any updates?

tarunsmalviya, 2019-05-24:
@jwhui Apologies for delay in response. 
Actually, I found that FTD is not connected with any other node, so I was not getting any response for **networkdiagnostic get** command.

jwhui, 2019-05-31:
@tarunsmalviya, thanks for the update.

closed 2019-05-31 by jwhui