openthread/openthread #3820: [efr32] Routing stopped once NCP ROLE changed to "end-device"

NickNamerr, 2019-05-15:
**Describe the bug**
Wpantund stopped to update ROUTING PARAMETERS on ROLE CHANGED to "end-device". Routing works only if ROLE of NCP is ROUTER.

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id 267a3b3
2. IEEE 802.15.4 hardware platform EFR32MG13p732GM48
3. Build steps https://github.com/openthread/openthread/tree/master/src/posix
4. Network topology 10 FTD

**Expected behavior**
In POSIX implementation it worked properly in Git commit id 2057a141ef75247197634d709bff4789d34c02d3

**Console/log output**
Thread:OffMeshRoutes = [
	"fdde::/64                  origin:user     stable:yes preference:medium  rloc:0x0000 next_hop_is_host:yes"
	"fdde::/64                  origin:ncp      stable:yes preference:medium  rloc:0x1400 next_hop_is_host:yes"
]
Thread:OnMeshPrefixes = [
	"fddd::                 prefix_len:64   origin:user     stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med] rloc:0x0000"
	"fddd::                 prefix_len:64   origin:ncp      stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med] rloc:0x1400"
]
Thread:Parent: <empty> (15335444)
Thread:PendingDataset = []
Thread:RLOC16 = **_0x0C06_**

**Additional context**
Add any other context about the problem here.

jwhui, 2019-05-15:
@NickNamerr, thanks for raising this issue.

If the NCP is configured as a router-capable FTD, then it is possible that the device is waiting to become a router. OpenThread currently has a check that avoids updating the network data when the device is pending an upgrade to router: https://github.com/openthread/openthread/blob/1a4b249a9527c88ea2f573fea9fda39bb025b5b1/src/core/thread/network_data_local.cpp#L356-L362

Can this be the root cause of your issue?

NickNamerr, 2019-05-16:
I have to debug it. BTW, the state changes to "end-device" and it keeps waiting(240 seconds?) till BR will become a "router". Is there a way, to prevent BR to change state so often? 
I observed such behavior during process of sending CoAP messages with payload size 1047 from Server to FTD over BR. In normal (idle) operations does not happen.

jwhui, 2019-05-16:
@NickNamerr, there are a couple reasons why a router may switch to "end-device":

1. The router is not receiving sequence number updates from the Leader and attempts to reattach to the network.

2. The router reset for some reason and did not receive an MLE Link Accept when it tries to reestablish its router role.

Also, to confirm, you noted that this behavior did not occur under the same test conditions with 2057a14?

Can you enable logging so that we can gain visibility into why the device is transitioning to "end-device" role?

NickNamerr, 2019-05-17:
@jwhui with 2057a14 it updates routing settings for POSIX<->OT_NCP_RADIO model, and traffic have continued to flow to destination and back.
How to enable LOGs for POSIX<->OT_NCP_RADIO ?

jwhui, 2019-05-17:
@NickNamerr, with 2057a14, did you observe the NCP changing state to "end-device"? I wonder if the state change to end-device is an indicator.

To enable logging in POSIX app, try add the following to `src/posix/platform/openthread-core-posix-config.h`:
```c
#define OPENTHREAD_CONFIG_LOG_LEVEL          OT_LOG_LEVEL_DEBG
#define OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE 4096
#define OPENTHREAD_CONFIG_LOG_SUFFIX         "\n"
```

NickNamerr, 2019-05-28:
@jwhui yes, I observed it. Sometimes it changes back to "router" , sometimes happens"Input/output error" and need to run again.

jwhui, 2019-05-29:
@NickNamerr, I'm wondering if the underlying reason for state change is due to the NCP resetting/hanging unexpectedly. Enabling logging and providing the log output could help diagnose the issue.

NickNamerr, 2019-05-31:
Is it enough to enable log only on POSIX app side? Do I need to do something for "ot-ncp-radio" app?

jwhui, 2019-05-31:
I think logs on the posix app side should give decent visibility to start. The log output will come via wpantund, so it will be good to see the wpantund and posix app syslog output.

jwhui, 2019-07-01:
@NickNamerr, any updates on this issue?

jwhui, 2019-07-12:
Closing stale issue.

closed 2019-07-12 by jwhui

NickNamerr, 2019-07-23:
@jwhui sorry, i did not have time to debug it yet, but i am going to do it.