openthread/openthread #3839: Dual-chip solution cannot deal with mesh local prefix change well

bukepo, 2019-05-20:
**Describe the bug**
Mesh Local Prefix is part of Thread network credential. However, if two notes are configured with different mesh local prefix but has the same master key. The second device will use its parent's mesh local prefix. This causes problem with Dual-chip solution because the new mesh local prefix is not notified to host before notifying the ip addresses, causing RLOC addresses being treated as non-mesh-local addresses. Thus, even wpantund was designed not adding RLOC addresses, they are actually added. What's worse is these addresses will be notified from host to NCP. Because the addresses are not of the new mesh-local prefix, they are treated addresses being added by user. As as result theses addresses will not be removed at the end.

**Expected behavior**
wpantund not add RLOC addresses.

abtink, 2019-05-20:
@bukepo , can you please help explain how the mesh-local addresses from NCP are being treated as addresses added by user? `wpantund` is designed to track origin of addresses (is there a bug related to this logic or something else is going on here...). Thanks.

abtink, 2019-05-20:
@bukepo , please ignore my earlier question.

jwhui, 2019-05-22:
Resolved by #3844.

closed 2019-05-22 by jwhui