openthread/openthread #3847: [mle] certification tests ED_6_1_6 and SED_6_1_6 fail

LuDuda, 2019-05-21:
I have verified that because of #3803, ED_6_1_6 and SED_6_1_6 fail.

The OT built from one previous commit successfully passes the certification test cases.

The device seems to attach to the first Router instead of sending another Parent Request with modified Scan Mask and attach to the REED.

Wireshark reference: [Test_6_1_6.zip](https://github.com/openthread/openthread/files/3201756/Test_6_1_6.zip)

We're investigating the issue, but help would be appreciated 😄

jwhui, 2019-05-21:
@LuDuda, thanks for identifying this issue!

I believe the issue is that the [check for Link Quality being 3](https://github.com/openthread/openthread/blob/master/src/core/thread/mle.cpp#L1641) is only for the receive direction and does not consider the bi-directional link quality.

I can submit a quick fix soon.

LuDuda, 2019-05-21:
Yes, it also aligns with my static code analysis.

Thank you for a fix proposal! I can verify it tomorrow.

PR #3850 (2019-05-22): [mle] fix bug in MLE attach behavior
Fixes a bug introduced in 5a7faa5.

A device performing MLE attach first sends an MLE Parent Request looking
for routers. If no MLE Parent Response is received from a neighboring router
where the link quality is 3 in both directions, the device shall send a
second MLE Parent Request that includes both routers and REEDs.

This commit adds the bi-directional link quality check to determine whether
a second MLE Parent Request should be sent.

Resolves #3847.
closed 2019-05-22 by jwhui