openthread/openthread #3852: Make pretty python scripts

bukepo, 2019-05-22:
**Is your feature request related to a problem? Please describe.**
A follow up to https://github.com/openthread/openthread/pull/3817.

**Describe the solution you'd like**
Add make pretty for python scripts and check that in Travis.

librasungirl, 2019-06-12:
For the code style, after some initial investigation,  a popular tool black (https://github.com/python/black) is chosen to auto format the THCI scripts (work with python 2.7) locally via below command
`python3 -m black -l 79 -S tools/harness-thci -t py27` (line length: 79, keep existing single quote in code)
(`black` could not format docstring or comment)

flake8 is chosen as the style consistency and source checker, however the length are relaxed in the configuration to be 119 for now due to legacy consideration (though [docformatter](https://pypi.org/project/docformatter/) could be used to update the docstring, I fails to find any tool to format comment, so here simply relax the rule)

Please review the style check sample in #3817. And feel free to comment if there is any different thoughts or suggestions. Thanks.

librasungirl, 2019-06-18:
@jwhui @bukepo thoughts?

bukepo, 2019-06-18:
I agree with @librasungirl 's proposal. And I think 119 line length is good.

PR #3951 (2019-07-11): [style] update python style to conform to PEP 8
With the exception of line length set to 119 vs. 79.

Add `tests/` and `tools/` to py-pretty-check.

Resolves #3852.
closed 2019-07-11 by jwhui