openthread/openthread #3866: Use static const for non-enumeration constants?

bukepo, 2019-05-24:
**Is your feature request related to a problem? Please describe.**
Currently OpenThread uses enum define constants event it is not an enumeration type. This causes issues like the compiler doesn't know the enumeration is signed or unsigned when choosing a overloaded methods.

**Describe the solution you'd like**
Use `static const <type>` to define non-enumeration constant, where `<type>` is a concrete numeric type, such as `int32_t`, `uint32_t`, `uint8_t` and so on.

**Additional context**
See #3865

KaSroka, 2019-06-15:
If we switch to C++11 we may specify a type for enum `enum asd : uint32_t { a = 123 };`

jwhui, 2021-07-21:
#6824 is starting to address this issue (by using `constexpr`.

jwhui, 2021-08-11:
Resolved by:

- #6824 
- #6845 
- #6875 
- #6895 
- #6907 
- #6909 
- #6910 
- #6911