openthread/openthread #3867: [cert] Leader_5_5_7 often fails

LuDuda, 2019-05-24:
I found today that the certification test `Leader_5_5_7` very often fails. I didn't notice that before since by default we have retries on each case - to eliminate e.g. sniffer packet loses etc..

This test contains 3x Routers and 1x Leader. Routers do not have connectivity with other routers.
In the middle of the test, Leader is turned off, and after >120s is back to the network. It then realizes that old partition is ceased, and performs MLE full attachment to one of the Router.

After Leader is powered-on again, Thread Harness test case waits for 60 seconds. After that, it checks if Leader attached to the network, and sent `Address Solicit` message. Unfortunately, the test sometimes fails due to the fact that `Address Solicit` message is not sent. I found that Leader (OT) sends this message after random time after attachment (in positive runs). I suspect it uses regular (0-120 router selection jitter), if it draws time between 0-60 then test passes, else it fails.

I checked `Router_5_5_7` test case where Leader (ARM) sends `Address Solicit` message right after attachment.

I'm not sure however if the bug is in OpenThread since I found in the specification only this:

> When a Router that has Child devices moves to a new Thread Network Partition, it immediately requests its original Router ID in the new Thread Network Partition. If the Router succeeds in getting the same Router ID assigned to it, the Child devices will stay attached. If it gets no Router ID or a different Router ID, then the Router clears its Child table. Its Child devices will eventually detect that their Parent link has failed and then reattach.

However, the Leader in this test case does not have any children attached to him.

RTT logs (from successul test case): [Leader_5_5_7_RTT_Logs_PASS.log](https://github.com/openthread/openthread/files/3216540/Leader_5_5_7_RTT_Logs_PASS.log)

From a code perspective, it seems we are in the `kAttachAny` mode and we don't call `BecomeRouter` function since `HasChildren()` returns 0 - [here](https://github.com/openthread/openthread/blob/master/src/core/thread/mle_router.cpp#L280)

If you don't find anything wrong in our implementation, I think we should file JIRA issue to either stop checking `Address Solicit` message or increase wait time from 60 to 120+ seconds in Thread Harness.

librasungirl, 2019-05-27:
@LuDuda , thanks for raising this issue. 

ARM Leader sends Address Solicit quickly because its THCI would reconfigure `routerselectionjitter` in `resetAndRejoin()` implementation.

OpenThread also introduced similar reconfiguration since #3524. 
https://github.com/openthread/openthread/blob/master/tools/harness-thci/OpenThread.py#L405
It should work for Leader DUT role. Would you please check if you are using the latest OpenThread.py and if it helps?

For Certification, agree with you that a JIRA should be filed as there should be no special requirement on routerselectionjitter setting on DUT device and at least 120s (maybe +5s) should be waited in the test script for robustness.

jwhui, 2019-05-28:
I agree that we should file a Thread Group JIRA. As mentioned, we should wait the full 120s + delta timeout for an Address Solicit message if one is expected.

LuDuda, 2019-05-28:
Thanks for your analysis and confirmation. I'll file a Thread Group JIRA.

librasungirl, 2019-06-03:
@LuDuda , do you have the JIRA number?

LuDuda, 2019-06-05:
Sure, [TESTPLAN-456](https://threadgroup.atlassian.net/browse/TESTPLAN-456).

jwhui, 2019-07-17:
This issue has been addressed in [TESTPLAN-456](https://threadgroup.atlassian.net/browse/TESTPLAN-456).

closed 2019-07-17 by jwhui