openthread/openthread #3874: Should we maintain Windows support?

KaSroka, 2019-05-28:
I would like to raise some discussion and hear your opinion.
There were no contributions to the Windows part of OpenThread for a long time. We are not testing if it's working, only if it builds. Sometimes it slows down the development of new features as we have to figure out how to make AppVeyor happy. Also, it adds some bloat to the code (OTAPI, OTCALL, WPP_NAME). My opinion is that we should remove support for Windows in OpenThread or we need a maintainer for this. Keeping it in the current state makes no sense.

abtink, 2019-05-28:
I agree...in favor of this.

jwhui, 2019-05-28:
Given that the last real contribution to the Windows platform was more than two years ago, I am fine with removing Windows support. Do you have proposed changes you'd like to submit? Or would you prefer I take an initial stab?

KaSroka, 2019-05-28:
@jwhui I took an initial stab but feel free to comment/contribute to my PR if you see more things that may be changed.

jwhui, 2019-06-06:
Resolved by #3878.

closed 2019-06-06 by jwhui