openthread/openthread #3877: Use numeric_limits for C++ code.

KaSroka, 2019-05-28:
C++ standard defines `std::numeric_limits` as a standard way to query various properties of arithmetic types instead of defines like `UINT32_MAX`. Some implementations of stdlib don't define those when building for C++, that's why we need things like [this](https://github.com/openthread/openthread/blob/master/include/openthread/platform/toolchain.h#L310). I think that we should use proper way defined by C++ standard with `std::numeric_limits<uint32_t>::max()` and `std::numeric_limits<uint8_t>::max()` instead. Of course only in C++ code. I've tested this change locally and code size doesn't change.

jwhui, 2019-05-28:
Using `std::numeric_limits` sounds like a good idea to me. Would you like to submit your proposed changes?

KaSroka, 2019-05-28:
Yes, I'll submit proposed changes after testing with other compilers.

KaSroka, 2019-05-29:
We would need to switch to C++11 from C++98 to make this work as pre C++11 version of `std::numeric_limits::max()` is not `constexpr` and we need this.

jwhui, 2019-05-29:
@KaSroka, thanks for pointing that out. I think moving to C++11 requires a separate issue for discussion, which I opened with #3880.

CiaranWoodward, 2019-05-30:
Am I right in saying that this would be the first use of the C++ Standard Library in all of openthread? In that case we might also need to rethink the "The use of the C++ Standard Library shall be avoided" Style Guide rule and consider what should/shouldn't be allowed.

jwhui, 2019-05-30:
@CiaranWoodward, good point. I think we should be selective in what portions of C++ Standard Library we allow. Maybe we can add a list of exceptions to the style guide, `std::numeric_limits` being one of the items?

jwhui, 2019-07-30:
Based on feedback in #3899, will not pursue C++11 for now.

closed 2019-07-30 by jwhui