openthread/openthread #3879: [time-sync] new devices joining a partition can cause the whole partition to lose synchronisation with the leader

josephn123, 2019-05-29:
**Describe the bug**

Device(s) holding a higher time sync sequence ID joins a partition. For example, the existing devices have just been been seeded by the leader with a time sync sequence of 10, and the new device(s) are seeding with a sequence ID of 136 as a relic from the previous partition. The existing devices, aside from the leader, accept and distribute the time synchronisation sequence, because they all consider it to be newer `(int8_t)(136 - 10) > 0` (see `time_sync_service.cpp` `HandleTimeSyncMessage()`). Take the case where the leader on the existing partition remains the same: none of the device(s) that just joined the partition are leaders. Now all devices aside from the leader have a time synchronisation based on the epoch of the previous partition, until the current leader seeds a time synchronisation with a time sequence ID greater than 136. In this case, with a time sync period of 30s, it will take 126 * 30 secs, i.e. 63 minutes before all the devices are in sync with the leader again. In the mean time, all devices aside from the leader will report `RESYNC_NEEDED` because nobody is actively seeding time sync with the epoch from the previous partition.

**To Reproduce**
Create a partition consisting of a few devices, their leader will begin seeding a time sync sequence from 0. In the mean time, configure another device and set `mTimeSyncSeq` in `time_sync_service.cpp` to be up to 127 greater than the current time sync sequence (use a dev board). This device will attach to the existing partition because it is bigger than the partition of the standalone device, so the leader shall remain the same. Observe the result.

**Expected behavior**
Perhaps upon joining a new partition, devices should reset their internal copy of `mTimeSyncSeq` to allow the leader on the partition they are joining to to seed them in all cases instead.

josephn123, 2019-05-29:
Another, more common scenario, is a recently booted device (therefore with a low sequence number i.e. 3) coming in range of a network currently seeding with a high sequence number (i.e. 131). The fresh device joins and all devices apart from the leader sync to the new device. Synchronisation to the leader is lost again for up to 63 minutes.

jwhui, 2019-05-29:
@josephn123, thank you for raising this issue.

I agree that a device attaching to a new partition should adopt whatever time sync sequence the partition has.

I also think it might be worthwhile for the Leader to reassert its authority by advancing its time sync sequence if it receives one that is ahead.

Thoughts?

/cc @chshu

chshu, 2019-05-29:
@josephn123 Thanks for reporting the issue. And I agree with you and Jonathon's suggestions, following changes may fix the issue:

1. Once a device attach a new partition, reset its mTimeSyncSeq to OT_TIME_SYNC_INVALID_SEQ (0), then it will accept any time sync sequence from the network.
2. If the Leader receives a newer time sync sequence (X), Leader updates it's mTimeSyncSeq to be (X + 1), so to keep Leader's time sync sequence always the newest. But keep the network time continuous (without jump).

@josephn123 Would you like to submit the fix?

josephn123, 2019-05-30:
Thank you both for the feedback and the suggestions. I will work on a corresponding fix.

PR #3884 (2019-05-31): [time-sync] reset sequence upon joining a new partition & ensure leader is always authoritative
Fixes #3879 - [time-sync] new devices joining a partition can cause the whole partition to lose synchronisation with the leader

Also added some logging under the 'core' channel. I'm not sure if this is quite right but I always find myself having to add temporary logging when working on this, so would be good to include some permanently.

Also removed setting `mLastTimeSyncReceived` to the current time in the constructor. I think could have introduced a bug where a device joining a non-time sync cap [... cut]
closed 2019-05-31 by jwhui