openthread/openthread #3880: Move language standard baseline to C++11?

jwhui, 2019-05-29:
The current language standard baseline is C++03.

Thoughts on whether we should move to supporting C++11?

One use case was raised in #3877.

abtink, 2019-05-29:
I am in favor in general. We should consider the list of the features that would be helpful (`auto`, ranged-`for` loops, `constexpr`,  `enum class`, etc.) and if there are any features we may want to limit the use of?
(new function syntax with `->` to specify return type, lambda with capture?)

KaSroka, 2019-05-29:
As C++ 20 is coming I think that we may consider newer standards as well (if they have features that may help us). Also we are building with std=gnu++98 configured currently, not 03. And the Android build have no standard defined (I cannot see any std= flag when running Android build on Travis)

KaSroka, 2019-05-29:
Here is a nice brief of new features in each version: https://github.com/AnthonyCalandra/modern-cpp-features

jwhui, 2019-05-30:
@KaSroka, we can consider adopting newer versions of the C++ language standard. At the same time, we should also consider compiler support as well. If there are good use cases for using newer language features that are well supported by compilers, we can discuss further.

KaSroka, 2019-05-31:
After looking into features of C++14 and 17 i cannot see any features that may help us maybe except relaxed requirements of constexpr in C++14 as well as no implicit const when using them.

gjc13, 2019-06-12:
Personally I found such code snippet to be appealing if we can adopt C++11 (which introduces argument pack in templates and std::forward):
```
#include <utility>

extern void LogInfo(const char *format, ...);

template <typename classname, typename U, typename RetType, typename ... ArgType>
class CtxBinder {
public:
    typedef RetType (classname::*Funct)(ArgType...);

    CtxBinder(U ctx, Funct f)
        :mctx(ctx), mf(f) {
    }

    RetType operator()(ArgType... args) {
       return (mctx->*mf)(std::forward<ArgType...>(args...));
    }

private:
    U mctx;
    Funct mf;
};

class A {
public:
    A();

    void f(int x);

    void trigger();

private:
    CtxBinder<A, A*, void, int> mf;
};

A::A() 
    :mf (this, &A::f) {
}

void A::f(int x) {
    LogInfo("%d\n", x);
}

void A::trigger() {
    mf(1);
}
```
A quick test showed that such pattern of binding `this` to member functions gives even smaller code size than currently used static function and context pattern. The static function code segment can be removed in the final -Os output. Such pattern also will reduce duplicate code only forwarding arguments to member functions.
However, `std::function` and lambda on the other hand, significantly increases the code size, which should be forbidden.

jwhui, 2019-07-30:
Based on feedback in #3899, will not pursue C++11 for now.

jwhui, 2020-06-05:
Based on new feedback https://github.com/openthread/openthread/pull/3899#issuecomment-639715510, reopening this issue again.

Please provide updated thoughts on whether we should move to supporting C++11.

PR #5077 (2020-06-16): [style] advance C++ language standard to C++11
Resolves #3880.
closed 2020-06-16 by jwhui