openthread/openthread #3896: add a size change report for pull request?

bukepo, 2019-06-04:
**Is your feature request related to a problem? Please describe.**
We need to get notified for unexpected size change. This is important for size restricted platforms. 

**Describe the solution you'd like**
Generate a size change report containing size changes of all platforms after running travis check.

**Describe alternatives you've considered**
None.

**Additional context**
None.

jwhui, 2019-06-04:
This would be very helpful.

The travis scripts print the output of `arm-none-eabi-size` in the build logs, but that is not visible enough.

Ideally, not only do we track difference for a given PR, but we also trend the code size so we can see the change over a longer period of time. It may be that not any one given change is significant, but many small changes are.

bukepo, 2019-08-19:
#4026 enabled reporting size diff for every PR. Next step is to record the history of size. I am running out of cycles recently. Anyone who want to contribute are welcome. The source code is at https://github.com/openthread/size-report

jwhui, 2020-02-20:
Resolved by #4026.

closed 2020-02-20 by jwhui