openthread/openthread #3934: nrf52840 radio not working on v0.9.3

gjc13, 2019-06-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior**
A clear and concise description of what you expected to happen.

**Console/log output**
If applicable, add console/log output to help explain your problem.

**Additional context**
Add any other context about the problem here.

gjc13, 2019-06-24:
@LuDuda friendly reminder

LuDuda, 2019-06-24:
Hello, @gjc13 we will take a look at this ASAP, however, note that this is engineering version of the silicon. Have you checked the production-ready silicon e.g. on PCA10056 v1.0.0 DK?

gjc13, 2019-06-24:
It works on v1.0.0 dongle. I don't have a v1.0 DK at my hand.

robertcragie, 2019-07-05:
I get the same problem on 0.9.0 and 0.9.2 boards using nRF52840 Q1AA:
```
> ipaddr
Done
> 
```
When I run `scan` I see no beacon request.

bukepo, 2019-07-09:
How about we revert v1.4.0 first? @LuDuda All our dev boards don't work now.

krlo, 2019-07-09:
nRF52840 Engineering A chip support is end of life.  Nordic has supported this Engineering version of silicon up to, and beyond, the previously communicated period, to support prototype builds with nRF52840.  In [this section](https://infocenter.nordicsemi.com/topic/comp_matrix_nrf52840/COMP/nrf52840/nrf52840_comp_matrix.html) you can find compatibility matrix between the nRF52840 IC revisions and development kits.

closed 2019-07-22 by jwhui