openthread/openthread #3956: A lot of logs is being dropped when using OPENTHREAD_CONFIG_LOG_OUTPUT_NCP_SPINEL

KaSroka, 2019-07-02:
**Describe the bug**

When using NCP with `OPENTHREAD_CONFIG_LOG_OUTPUT_NCP_SPINEL` and debug log level we drop a lot of logs. It's transport independent - the same happens on UART, USB or SPI. And the transport is not the bottleneck - under high load I've measured 8MHz SPI usage of ~4%.

I've also added some counters to measure a number of dropped logs. Here is an example with 115200 baud rate UART, the board was alone in the network with no traffic at all (idle):
```bash
"NCP:Counter:LogReqestCount       = 1857"
"NCP:Counter:LogReqestFail        = 811"
```
This shows that almost 50% of logs are dropped during idle state, under load we have seen numbers as high as 94%

**To Reproduce**

Build NCP with `OPENTHREAD_CONFIG_LOG_OUTPUT_NCP_SPINEL` and debug log level.

**Expected behavior**

All logs are logged successfully until the transport medium is saturated.

**Console/log output**

Sending an advertisement alone results in `NOMEM` status.
```bash
wpantund[26272]: [NCP->] (13) PROP_VALUE_IS(LAST_STATUS) [00]
wpantund[26272]: [-NCP-]: Last status (OK, 0)
wpantund[26272]: NCP => [INFO]-MLE-----: -MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
wpantund[26272]: NCP => [DEBG]-MAC-----: -MAC-----: Request to start operation "TransmitData"
wpantund[26272]: NCP => [DEBG]-MAC-----: -MAC-----: Starting operation "TransmitData"
wpantund[26272]: NCP => [DEBG]-MAC-----: -MAC-----: RadioState: Receive -> CsmaBackoff
wpantund[26272]: NCP => [DEBG]-MAC-----: -MAC-----: RadioState: CsmaBackoff -> Transmit
wpantund[26272]: NCP => [DEBG]-MAC-----: -MAC-----: RadioState: Transmit -> Receive
wpantund[26272]: NCP => [DEBG]-MAC-----: ==============================[TX len=069]===============================
wpantund[26272]: NCP => [DEBG]-MAC-----: | 41 D8 C0 CD AB FF FF EA | FE 5D A3 01 F3 8F E6 7F | AX@M+..j~]#.s.f.
wpantund[26272]: NCP => [DEBG]-MAC-----: | 3B 01 F0 4D 4C 4D 4C C0 | 5B 00 15 E8 00 00 00 00 | ;.pMLML@[..h....
wpantund[26272]: [NCP->] (0) PROP_VALUE_IS(LAST_STATUS) [0B]
wpantund[26272]: [-NCP-]: Last status (NOMEM, 11)
```
When sending an advertisement I can see following figures:
```
"NCP:Counter:LogReqestCount       = 18"
"NCP:Counter:LogReqestFail        = 9"
```
This means that from 18 log attempts only 9 were successfull (and we can see 9 log lines above).

Tested on RPi and Ubuntu PC.

abtink, 2019-07-02:
What is your NCP buffer size config? Would recommend increasing it if you are using default.
`OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE`
We use more than 3500 typically (specially when logs are enabled and go through NCP).

KaSroka, 2019-07-03:
@abtink above measurement was done with default 512B and 115200 baud rate. I've changed this to 4096B but still can see some logs dropped. This is the result during commissioning:
```bash
NCP:Counter:AllSpinel = [
	"NCP:Counter:LogReqestCount       = 704"
	"NCP:Counter:LogReqestFail        = 325"
	"NCP:Counter:EnqueueResponseCount = 5"
	"NCP:Counter:EnqueueResponseFail  = 0"
	"NCP:Counter:UpdatePropsCount     = 56"
	"NCP:Counter:UpdatePropsFail      = 11"
	"NCP:Counter:SendDatagramCount    = 0"
	"NCP:Counter:SendDatagramFail     = 0"
	"NCP:Counter:SendLogByteCount     = 45732"
	"NCP:Counter:DroppedLogByteCount  = 21052"
]
```
~50% is dropped despite 4kB buffer size.

With 16kB buffer:
```bash
NCP:Counter:AllSpinel = [
	"NCP:Counter:LogReqestCount       = 685"
	"NCP:Counter:LogReqestFail        = 142"
	"NCP:Counter:EnqueueResponseCount = 4"
	"NCP:Counter:EnqueueResponseFail  = 0"
	"NCP:Counter:UpdatePropsCount     = 28"
	"NCP:Counter:UpdatePropsFail      = 3"
	"NCP:Counter:SendDatagramCount    = 0"
	"NCP:Counter:SendDatagramFail     = 0"
	"NCP:Counter:SendLogByteCount     = 44516"
	"NCP:Counter:DroppedLogByteCount  = 8999"
]
```
~20% is dropped despite 16kB buffer size.

When switched to 1000000 baud and 16k buffer I can see further improvement - when pinging with 1200B payload every 1s I get 10% of dropped logs - but the UART usage is around 54% - far from saturated. Should I use even bigger buffer here?

abtink, 2019-07-03:
@KaSroka , if you can use larger buffers go for it. Higher UART buaud rate may also help.
One other suggestion is to reduce the log level. I notice you are using log level `DEBG`. That's generally very verbose. We commonly use `INFO` (or `NOTE`) with ~3.5 KB buffer (on dev builds) and generally works ok (rarely we see dropped logs). The `DEBG` we use for debugging only.

KaSroka, 2019-07-08:
@abtink Thanks for suggestions, bigger buffer helped.

closed 2019-07-08 by KaSroka