openthread/openthread #3957: Diag power feature is not used anywhere

KaSroka, 2019-07-03:
`diag power` is not used by any platform expect nRF52840/nRF52811 (we have a plan to change this behavior) and it doesn't result in power change for any platform. I think that this command should be removed or call `otPlatRadioSetTransmitPower` for all platforms. In the current state, its behaviour is misleading. What do you think?

chshu, 2019-07-09:
The txpower was set within each package before, the feature was removed since #2352, then the diag txpower is maintained under platform layer by calling otPlatDiagTxPowerSet(), so the "diag power" cmd is still useful.

KaSroka, 2019-07-10:
Yes but all `otPlatDiagTxPowerSet` implementations are empty except Nordic ones. #3969 fixes behaviour of `diag power`.

jwhui, 2019-07-30:
@KaSroka, I agree that it probably makes sense to remove the `otPlatRadioSetTransmitPower` API since it is not an API required/used by OpenThread core. We can always leave power set for the purpose of factory diags as a platform-specific CLI.

jwhui, 2019-07-30:
@KaSroka, would you be willing to submit a PR with the proposed change?

abtink, 2019-07-30:
@KaSroka , @jwhui , the `otPlatRadioSetTransmitPower` API is not used by core but it's used by ncp code and is exposed as spinel property `SPINEL_PROP_PHY_TX_POWER` which lets the power to be set from driver (`wpantund`). This let the higher layer code to control power in NCP builds where diags feature may be fully disabled. I suggest we keep it for now. 

Thoughts?

jwhui, 2019-08-06:
@abtink, thanks for pointing that out. I'm fine to keep `otPlatRadioSetTransmitPower()` for now.

KaSroka, 2019-08-06:
Change from #3969 is enough in my opinion. Now we have two separate commands - one for normal operation and one for diag. We may close this issue if you agree.

jwhui, 2019-08-06:
@KaSroka, sounds good. Thanks!

closed 2019-08-06 by jwhui