openthread/openthread #3961: Undefined reference to Frame::ToInfoString function when log level < LOG_LEVEL_INFO

marvensilabs, 2019-07-03:
Building OpenThread I see the linker fails if the OPENTHREAD_CONFIG_LOG_LEVEL is set to OT_LOG_LEVEL_NOTE.  However, the link succeeds if it is set to LOG_LEVEL_INFO.

The commit ID used for testing is  0d75bbe.

The error message I see shows:
/examples/../src/core/thread/mesh_forwarder.cpp:1774: undefined reference to `ot::Mac::Frame::ToInfoString() const'

For building I've used the efr32mg12 platform and added 
 -DOPENTHREAD_CONFIG_LOG_LEVEL=OT_LOG_LEVEL_NOTE to the COMMONCFLAGS section of the makefile and built with:

$ make -f examples/Makefile-efr32mg12 COMMISSIONER=1 JOINER=1 DHCP6_CLIENT=1 DHCP6_SERVER=1 BOARD=BRD4161A

It looks like Frame::ToInfoString in is only defined in mac_frame.cpp if log level >= OT_LOG_LEVEL_INFO and OPENTHREAD_CONFIG_LOG_MAC is set to 1.

However this function is used in calls to otLogNoteMac macros in mesh_forwarder.cpp.

Should these calls be converted to otLogInfoMac or should the ifdef in mac_frame.cpp be changed to OT_LOG_LEVEL_NOTE?

jwhui, 2019-07-03:
@marvensilabs, thank you for raising this issue!

I believe we should change the ifdef in `mac_frame.cpp` to `OT_LOG_LEVEL_NOTE`. I'll submit a fix shortly.

jwhui, 2019-07-03:
@marvensilabs, just submitted #3962. Please review.

PR #3962 (2019-07-09): [logs] compile Mac::Frame::ToInfoString at NOTE log level
This fixes a linker error when buiding with OT_LOG_LEVEL_NOTE.

MeshForwarder depends on Mac::Frame::ToInfoString at the NOTE level.

Resolves #3961.
closed 2019-07-09 by jwhui