openthread/openthread #3990: Enrollment over Secure Transport (EST) client over CoAP Secure

scnmzhaw, 2019-07-12:
I've implemented an EST client for my fork of the OpenThread stack for a project and would like to share it with you, if that's something you are interested in. Unfortunatly, managing an EST server is too much to handle for a constrained device, so i can't provide an example with two nodes acting as server and client, so i provided an IPv6 address of an EST server which my team set up for our testing. I would like to see this as a frist step for providing an open source soultion for security in the IoT for which no proprietary hardware is necessary. 

So what do you think? Is this something you are interested in?

wgtdkp, 2019-10-02:
Thanks @AuSchell, we are glad to see that there is someone outside of OpenThread interested in implementing the Thread 1.2 CCM-related feature on top of OpenThread (**Are you implementing the EST client conforming to the Thread 1.2 Commercial Commissioning Extension?**). Actually, we have been developing the Thread 1.2 Commercial Commissioning Extension for some time. But, there are still some undecided problems of the spec and tests (Thread harness) before we can open the implementation. 

**If you are truly interested in the CCM feature, we may consider to open source our CCM implementation before getting the Thread certificate and continue developing CCM feature in a more open way**.

AuSchell, 2019-10-02:
Thanks @wgtdkp for your response, we are very interested in that. We have an implementation which is runnig most parts of ANIMA/BRSKI/EST protocols, but only on an application level and in a more general approach for proof of concept than specificly for Thread. Here is a link to the paper we released about that: [https://doi.org/10.21256/zhaw-2750](https://doi.org/10.21256/zhaw-2750).

So if we can collaborate to implement the CCM features, that would be great.

wgtdkp, 2019-10-08:
Hi @AuSchell, Do you have access to Thread 1.2 specifications?

AuSchell, 2019-10-08:
Hello @wgtdkp, our institute is applying for an academic membership to the Thread Group, our legal department is currently evaluating the conditions.

wgtdkp, 2019-10-09:
That's great, because one will need the newest Thread 1.2 spec to develop the CCM feature. Considering more and more Thread users and companies are switching to OpenThread, it's better for us to take advantage of the OpenThread community for new feature development. So, we will discuss about speeding open source of our efforts on CCM feature, stay tunned!

jwhui, 2020-06-05:
This will be combined into the Thread 1.2 effort.

closed 2020-06-05 by jwhui