openthread/openthread #4013: Question about operating as a SED

nzinn08, 2019-07-21:
Since I couldn't find a tutorial about how to initialize a device as a sleepy-end device anywhere on the OpenThread website I wanted to verify with the people here that everything I want to do for SED initialization is correct from what I've researched.

For APIs to use:
1. otThreadSetLinkMode to only have Secure IEEE 802.15.4 data requests enabled thereby disabling RX on when idle. 
2. otLinkSendDataRequest when my MCU wakes up from it's own low-power mode in order to poll for messages from its parent.

Are these all the APIs I would need?

Related questions: 
1. For the Rx on when idle parameter, what is defined to be idle? When will the receiver be on and when will it be off?
2. For the message received after the otLinkSendDataRequest, how/where would I check the frame pending bit?
3. How does the parent know whether to do direct or indirect transmission as shown here https://openthread.io/guides/porting/implement-advanced-features?authuser=1 ? Is this from the Rx on when idle parameter?
4. How does otSetChildTimeout work? Would the timeout continue if my MCU core is disabled while in low-power mode?

Thanks in advance and sorry if this is all explained somewhere and I just can't find it.

jwhui, 2019-07-22:
To enable Sleepy End Device (SED) operation, use [otThreadSetLinkMode](https://openthread.io/reference/group/api-thread-general#otthreadsetlinkmode) with the following config:
- mDeviceType = 0
- mNetworkData = 0 (but could be 1 if you want all network data changes)
- mRxOnWhenIdle = 0
- mSecureDataRequests = 1

To receive messages, a SED needs to poll its parent by sending an IEEE 802.15.4 MAC Data Request message. By default, OpenThread manages this transmission with the following configurations:
- [otThreadSetChildTimeout](https://openthread.io/reference/group/api-thread-general#otthreadsetchildtimeout) - by default, OpenThread will send MAC Data Request messages to maintain the child timeout value.
- [otLinkSetPollPeriod](https://openthread.io/reference/group/api-link-link#otlinksetpollperiod) - to configure a period that is different from the default.
- [otLinkSendDataRequest](https://openthread.io/reference/group/api-link-link#otlinksenddatarequest) - to trigger an immediate IEEE 802.15.4 MAC Data Request transmission.

To answer your questions above:
1. When attached to a Parent, a SED's radio is considered idle when it is not actively transmitting (sending data to Parent) or trying to receive via an IEEE 802.15.4 Data Request transmission.
1. OpenThread handles the IEEE 802.15.4 Indirect Transmission process, including the generation/processing of the Frame Pending bit.
1. A Thread Parent relies on the Link Mode configuration from its child to determine whether transmissions should be sent via direct or indirect transmissions.
1. OpenThread requires a platform to provide a millisecond alarm. OpenThread uses this alarm to maintain time. In other words, the MCU is responsible for maintaining the millisecond alarm while it is in a low-power state.

Hope that helps.

nzinn08, 2019-07-22:
Thanks for the detailed reply @jwhui! This definitely helps. I just have a few follow-up questions now.

1. In your reply to my original question 2, you stated that OpenThread handles this processing of the frame pending bit. The problem here is that I want to trigger an immediate Data Request and if there is no data pending then I'd like to immediately have my MCU enter low power mode. Is there any way to check this frame pending bit manually?
2. Is the child timeout necessary to be implemented or can it be disabled? (This kinda relates to the next question)
3. What makes us become disconnected from an OpenThread network? Is it the child timeout? Would shutting off our radio make us become disconnected from the network and cause us to have to reauthenticate and reconnect?

jwhui, 2019-07-22:
To answer your questions:
1. Power management should be handled at the platform layer. For example, if there are no platform-specific events pending, the platform can go to sleep until the next event (e.g. alarm timer fire).
1. Thread specifies the use of a child timeout parameter to help the Parent determine when it can time out state about a child. The child timeout is communicated by the Child to the Parent. If the Parent has not received a keep-alive message (e.g. Data Request) within the child timeout, the Parent will remove state about the Child.
1. For a sleepy end device, the child timeout is the primary mechanism for disconnection from the Parent. As mentioned above, if the child fails to send a keep-alive message to its parent within the child timeout, the Parent will consider the child detached.

nzinn08, 2019-07-24:
Thanks for all the help, I really appreciate it!

closed 2019-07-24 by nzinn08