openthread/openthread #4018: [efr32] FTD role becames "DETACHED" after "ROUTER"

NickNamerr, 2019-07-23:
**Describe the bug**
After a while FTD goes to role DETACHED.

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id : db8c4c83399e5df7514d2ef8d4a57423e5f1b39a
2. IEEE 802.15.4 hardware platform : efr32mg12p432f1024gm48
3. Build steps : make -f examples/Makefile-efr32mg12 SERVICE=1 COAP=1 BOARD=BRD4170A UDP_FORWARD=1 JOINER=1
4. Network topology: 4 FTD, 15 MTD(SED), 1 BR

**Expected behavior**
FTD must reattach to the network.

**Console/log output**
> state
detached
Done
> bufferinfo
total: 44
free: 2
6lo send: 0 0
6lo reas: 0 0
ip6: 0 0
mpl: 21 42
mle: 0 0
arp: 0 0
coap: 0 0
coap secure: 0 0
application coap: 0 0
Done
>

**Additional context**
Updated from commit id "a07751ec0d7b02394670803e40dd080833d94b36", before never seen this behavior.

jwhui, 2019-07-23:
@NickNamerr, thanks for raising this issue.

I have not seen this behavior before, where the message buffers are getting "stuck" in the MPL queue.

Can you check if [Mpl::HandleRetransmissionTimer](https://github.com/openthread/openthread/blob/bcbe8e2179e0fece8772c9a2442f1cd4f698a88f/src/core/net/ip6_mpl.cpp#L334) is getting called? If so, you should see calls to [mBufferedMessageSet.Dequeue()](https://github.com/openthread/openthread/blob/master/src/core/net/ip6_mpl.cpp#L392).

Also, it would help if you could provide more information about the application and the traffic that it generates.

NickNamerr, 2019-07-23:
@jwhui Yes, I can do it. Only the question, when shall I do it? Because device goes to role DETTACHED unpredictably. I could debug it only when detect that there not traffic comes from devices.

>> Also, it would help if you could provide more information about the application and the traffic that it generates.

I think MTD (SED) does not poll messages , but have to check it. BTW - I believe there shall be a logic, to handle this cases.

jwhui, 2019-07-25:
@NickNamerr, does the device get stuck in the DETACHED state? Or does it eventually recover and reattach? I would like to know if message buffers are getting stuck in the MPL queue, or is it transient behavior?

Note that MPL is not used to deliver messages to MTDs. The buffer info that you provided above does not indicate any other holder of message buffers.

In my previous question, I was looking to see if you had an application that is generating any traffic on top of what OpenThread generates. Or is this simply OpenThread running without application traffic?

NickNamerr, 2019-07-26:
@jwhui device does not get stuck (MCU freeze?), it remains in DETACHED state with FW running, cli works fine. I never awaited moment when device eventually recovers and reattaches (stays for 7-13 hours in DETACHED state, while other devices have worked fine). 

There are periodically CoAP messages form MTD/FTD to server through border router and back.

The symptoms is following:
1) MPL buffers number begins to increase. As much child in "child list" as faster it increasing.
2) Once it reached 21 42 - within timeout is going to role DETACHED

NickNamerr, 2019-07-29:
@jwhui sorry, accidentally pressed "close and comment", reopened it.

hemanth-silabs, 2019-08-07:
@marvensilabs @nigelg-silabs have a look at the issue above.

marvensilabs, 2019-08-07:
We have been looking into a similar issue where we run out of buffers.
From testing it looks like the alarm is not being set if the expiration time t0 + dt is too small or ends up being before the time the RAIL_SetTimer is called.  This then prevents any further alarms from being set.  We have a fix that we are currently testing for this and should fix the issue described here.

jwhui, 2019-08-14:
Resolved by #4081.

closed 2019-08-14 by jwhui