openthread/openthread #4022: Complete out-of-tree OpenThread build

rlubos, 2019-07-24:
I've been working recently on improving Zephyr/OpenThread integration at a build system level, and I faced a problem with complete out-of-tree build of OpenThread.

My goal is to build OpenThread completely out-of-tree (from Zephyr project build directory) so that the original repository remains intact. While it works perfectly fine to call the `configure` script and build the libraries from the external directory, I could not find a way to initiate the autotools with a `bootstrap` script from the outside. Is that even possible to call the `bootstrap` from the external directory, so that it produces any output into that directory? I'm not sure if it's the autotools limitation or do I simply miss something.

jwhui, 2019-07-25:
@rlubos, unfortunately, I am not aware of any way to perform `./bootstrap` out-of-tree.

rlubos, 2019-08-07:
Thank you for the answer, I suppose we can close the issue then.

closed 2019-08-07 by rlubos