openthread/openthread #4025: panid command produces different output format than it takes as argument

shalak, 2019-07-25:
**Describe the bug**
When used to set configuration, the `panid` CLI command converts decimal input into hexadecimal, but when used to get configuration it returns hexadecimal value, with no `0x` prefix - thus causing ambiguity.

**To Reproduce**
Use `panid` command in CLI:

```
> panid 1234
Done
> panid
04d2
Done
```

**Expected behavior**
AC: format is the same for both input and output of `panid` command, by implicitly adding `0x` either to input or output.
```
> panid 1234
Done
> panid
0x1234
Done
```
OR

```
> panid 1234
Done
> panid
0x04d2
Done
```
**Additional context**
Basically what should be expected is a possibility to feed output of `panid` on one device to input of `panid` on another device and they both configure the same panid.

jwhui, 2019-07-25:
@shalak, thanks for raising this issue.

I agree that the input and output should be consistent. How about updating the output to always prefix the `0x` in front? If you agree, would you like to submit a PR for that?

shalak, 2019-07-26:
Will [that](https://github.com/openthread/openthread/pull/4028) solve the issue?

jwhui, 2019-07-26:
Resolved by #4028.

closed 2019-07-26 by jwhui