openthread/openthread #4036: Minimum set of information to attach to a network

bstraehl, 2019-07-30:
The API description states that **"The minimum set of information needed to attach is the PAN ID and Thread Master Key."**

But that's not true. Here the initialisation of a leader and an end device using CLI commands:
For the leader:
```
� reset
> dataset init new
Done
> dataset commit active
Done
> dataset active
Active Timestamp: 1
Channel: 15
Channel Mask: 07fff800
Ext PAN ID: ab6f0f99db116ec3
Mesh Local Prefix: fd19:ea13:f1fa:e501/64
Master Key: 1222558b01070cb6211ae2ad888f9429
Network Name: OpenThread-096e
PAN ID: 0x096e
PSKc: c2d539fc2f64b8b56713de1de2f6de8f
Security Policy: 0, onrcb
Done
> ifconfig up
Done
> thread start
Done
> state
detached
Done
> state
detached
Done
> state
leader
Done
```
And the slave:
```
� reset
> dataset clear
Done
> dataset masterkey 1222558b01070cb6211ae2ad888f9429
Done
> dataset commit active
Done
> dataset active
Master Key: 1222558b01070cb6211ae2ad888f9429
Done
> ifconfig up
Done
> thread start
Done
> state
detached
Done
> state
detached
Done
> state
detached
Done
> state
detached
Done
> state
child
Done
```
Even worse. If the end device is configured with a wrong PANID, as long as the master key is the same as the leader has, it's possible to attach to the network:
```
� reset
> dataset clear
Done
> dataset masterkey 1222558b01070cb6211ae2ad888f9429
Done
> dataset panid 0xbeef
Done
> dataset commit active
Done
> dataset active
Master Key: 1222558b01070cb6211ae2ad888f9429
PAN ID: 0xbeef
Done
> ifconfig up
Done
> thread start
Done
> state
detached
Done
> state
detached
Done
> state
detached
Done
> state
detached
Done
> state
detached
Done
> state
child
Done
```

jwhui, 2019-07-30:
@bstraehl, I suspect that your device was not in a "factory reset" state. After a reset, if the device has an Active Dataset in non-volatile storage, it will restore settings including the PAN ID.

You can check this by using the `panid` CLI command. When booting from factory reset, the PAN ID should show as `0xffff`.

In place of the `reset` command, try issuing the `factoryreset` command instead.

bstraehl, 2019-07-31:
@jwhui As you have seen in my second end device configuration, I redefined the PAN ID to 0xbeef, which is not the correct value, if you compare with the leaders configuration. Then I committed this to the active dataset. So I assume, that no matter if there was a factory reset or not, that this new value will be used during the next **thread start**, which is not the case. The PAN ID is simply ignored. 
However, as you suggested, I tried with a factory reset before:
```
factoryreset
> dataset active
Error 23: NotFound
> panid
ffff
Done
```
As you stated, the panid id 0xffff now. But then I configured the end device again with a wrong PAN ID (0xabcd this time) and as you can see it still becomes a child of a network with PAD ID 0x096e. So I still believe, that the PAN ID is ignored.
```
> factoryreset
> dataset active 
Error 23: NotFound
> panid 
ffff
Done
> dataset masterkey 1222558b01070cb6211ae2ad888f9429
Done
> dataset panid 0xabcd
Done
> dataset commit active
Done
> dataset active
Master Key: 1222558b01070cb6211ae2ad888f9429
PAN ID: 0xabcd
Done
> ifconfig up
Done
> thread start
Done
> state
detached
Done
> state
detached
Done
> state
detached
Done
> state
child
Done
> dataset active
Active Timestamp: 1
Channel: 13
Channel Mask: 07fff800
Ext PAN ID: b5bc1180bdcc0aec
Mesh Local Prefix: fd80:6544:2547:3555/64
Master Key: 1222558b01070cb6211ae2ad888f9429
Network Name: OpenThread-72ac
PAN ID: 0x096e
PSKc: 8c86011574e5f35db8201078a5d025f4
Security Policy: 0, onrcb
Done
```

jwhui, 2019-07-31:
@bstraehl, yes, the documentation is incorrect! Thanks for catching that.

As you noted, only the master key is required to join a network. As specified in [Thread 1.1.1 Section 8.7.7](https://www.threadgroup.org/support#specifications), MLE Announce messages may be used to resolve differences in Channel and/or PAN ID configuration.

I'll submit a PR soon to fix the docs and allow starting Thread without explicitly configuring the PAN ID.

bstraehl, 2019-07-31:
@jwhui Thank you for the clarification. But then also this "[join an existing network](https://openthread.io/guides/thread-primer/network-discovery#join_an_existing_network)" is wrong and I tried it. If you know PAN ID, extended PAN ID, channel and network name, this will not be sufficient to join a network, because you also need to know the master key. And the other way round, if you know the master key, you don't need to know anything else. All what is necessary to join a network is the master key. Every other parameter of the child's active dataset will be updated with data from the leader. Right?

jwhui, 2019-07-31:
@bstraehl, yes that page should be updated as well to include the master key. While all that is necessary to attach to a network is the master key, knowing the channel and PAN ID helps reduce attaching delays/transmissions since those parameters do not need to be discovered.

bstraehl, 2019-07-31:
@jwhui Thank you.

PR #4051 (2019-08-05): [dataset] only master key is needed to attach
Resolves #4036
closed 2019-07-31 by bstraehl