openthread/openthread #4038: [doc] build switches

LuDuda, 2019-07-30:
**Is your feature request related to a problem? Please describe.**
I have a feeling that apart of OT Core developers, it is hard for OT users to know how and which switches to set in the building process. It is also confirmed by the number of support questions we receive.

**Describe the solution you'd like**
Creating the `README.md` file inside `openthread/examples` directory, that would describe all the switches, their usage, and default values. Additionally, the short description of extra feature for a specific platform such as `nRF52840` could be listed there in a separate section - or alternatively, extra flags could be listed within the `README.md` of the platform directory, but with unified form (for each platform) - for user convenience.

Such documentation should clearly describe that the final configuration is in `openthread-core-config.h` and switches just enable/disable the commonly used modules/functionalities.

If this is something desirable, I can try to prepare some initial draft when I'll find a time. However it is nothing critical, I guess.

**Describe alternatives you've considered**
Don't change the current approach and improve documentation inside the platform's only, documentation on OpenThread.io, implementing something like `menuconfig` and remove common-switches... 

**Describe the ideal, beautiful world**
Table with description of each common-switch should contain information about how much RAM/ROM size their add compared to the default build without any flags e.g. for nRF52840.

jwhui, 2019-07-30:
@LuDuda, sounds like a great idea!

piotrkoziar, 2020-09-02:
Hello, I have just created draft PR. Please take a look.
https://github.com/openthread/openthread/pull/5484

closed 2020-12-22 by jwhui