openthread/openthread #4039: Question about sending UDP Message to SED

nzinn08, 2019-07-30:
I am attempting to receive a UDP Message from a SED on a mesh network. I have set up the SED with link mode of "s" and it successfully joins the in-place network as a child of the leader. I then have a FTD join the network and switch to a router and is also a child of the leader. I open and bind a UDP socket for the SED and send a mesh-local UDP broadcast message to that port from the FTD and the callback never gets called from the SED side. I have tried continuously calling otLinkSendDataRequest() as well as setting a Poll Period for the SED and nothing works. 

I have verified that the callback does get called when I set the link mode on the SED to "rs" so I don't believe the UDP socket is the issue.

Is there something else I need to be doing in order for a SED to receive a mesh-local broadcast message through UDP?

jwhui, 2019-07-30:
Are you using `ff03::1` as the IPv6 destination address? If so, that will not be forwarded to a Sleepy End Device (SED). The only IPv6 multicast addresses that a SED implicitly registers are the Link-Local All Thread Nodes and Mesh-Local All Thread Nodes addreses. Both are Unicast Prefix-Based Multicast Addresses [RFC 3306](https://tools.ietf.org/html/rfc3306). See [Thread 1.1.1 Section 5.2.3](https://www.threadgroup.org/support#specifications) for more details.

Couple clarifications:

1. A Thread device only operates in one of Child, Router, Leader roles at a time. A Thread device cannot be both a child and router at the same time.

1. IPv6 does not support a "broadcast" primitive, but instead uses "multicast" for addressing a group of devices.

Hope that helps.

nzinn08, 2019-07-30:
Ooo thanks for the link to the Thread spec, can't believe I didn't look at that before! So is the mesh-local prefix always the upper 64 bits of the RLOC for every device on the same thread network? Also, is the mesh-local prefix randomly generated upon creation of a new network?

nzinn08, 2019-07-31:
Thanks for the help!

closed 2019-07-31 by nzinn08