openthread/openthread #4061: Macro update message in openthread-core-config-check.h misleading

gjc13, 2019-08-06:
For instance we have message:
```
#error "OPENTHREAD_CONFIG_ENABLE_ATTACH_BACKOFF was replaced by OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_JITTER_INTERVAL."
```
However we are using `OPENTHREAD_CONFIG_MLE_ATTACH_BACKOFF_ENABLE` to replace it. Such misleading messages make it hard and dangerous to update the product configs using OpenThread.

Another weird message is:
```
#error "OPENTHREAD_CONFIG_NUM_SLAAC_ADDRESSES was replaced by NRF_802154_PENDING_EXTENDED_ADDRESSES.
```
NRF_XXX macros should definitely not appear here.

I've not gone through the whole file to check which macro replacement messages are questionable so the above macros are just examples.

jwhui, 2019-08-06:
@gjc13, thanks for catching this issue. Submitted #4062.

abtink, 2019-08-06:
@gjc13 we decided to add the check macros after the main config change was done by Jonathan.
I argued for it (and added it) since I knew it would make the task of changing the config on internal projects easier.

Wrote a script to try to search and match the old config name to the new one. The script tried to match as many words as possible. Obviously there would be cases like above which the script would make a mistake. 

The key point for this was to make sure the old names are not being used and give an error if any of old the names were still defined. The error message is intended to give a hint to what the new name may be.

PR #4067 (2019-08-09): [config] fix config check helper messages
This commit updates helper error message in `core-config-check.h`
file. It also adds a comment to indicate that error message is
intended as a hint to what the new configuration parameter name
may be.

Resolves #4061.
closed 2019-08-09 by jwhui