openthread/openthread #4074: Running thread-cert tests on a SOC

bstraehl, 2019-08-08:
I'm trying to run thread-cert tests on SOCs, which are connected over UARTs to a PC. So I was doing something like that:
```NODE_TYPE=soc top_builddir=../../.. python Cert_6_3_01_OrphanReattach.py```

This threw the error:
```======================================================================
ERROR: test (__main__.Cert_5_1_02_ChildAddressTimeout)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "Cert_5_1_02_ChildAddressTimeout.py", line 50, in setUp
    self.nodes[i] = node.Node(i, (i in MTDS), simulator=self.simulator)
  File "/home/slb@simdev.defence.ruag.com/projects/pan/pan/openthread/tests/scripts/thread-cert/node.py", line 55, in __init__
    self.__init_soc(nodeid)
  File "/home/slb@simdev.defence.ruag.com/projects/pan/pan/openthread/tests/scripts/thread-cert/node.py", line 149, in __init_soc
    import fdpexpect
ImportError: No module named fdpexpect
```
I fixed this by:
```
diff --git a/openthread/tests/scripts/thread-cert/node.py b/openthread/tests/scripts/thread-cert/node.py
index c484d9dd..0adff6e3 100755
--- a/openthread/tests/scripts/thread-cert/node.py
+++ b/openthread/tests/scripts/thread-cert/node.py
@@ -146,9 +146,9 @@ class Node:
 
     def __init_soc(self, nodeid):
         """ Initialize a System-on-a-chip node connected via UART. """
-        import fdpexpect
+        import pexpect.fdpexpect as fdpexpect
 
-        serialPort = '/dev/ttyUSB%d' % ((nodeid - 1) * 2)
+        serialPort = '/dev/ttyUSB%d' % (nodeid - 1)
         self.pexpect = fdpexpect.fdspawn(
             os.open(serialPort, os.O_RDWR | os.O_NONBLOCK | os.O_NOCTTY)
         )
```
I did the second modification in the diff above, because I've seen that (nodeid-1) is multiplied by two, which I don't understand, because the name of the interfaces to be used, are derived from that. My devices are attached to /dev/ttyUSB[0-2], while the original code tries to use /dev/ttyUSB0, /dev/ttyUSB2 and /dev/ttyUSB4. With my modifications everything works. Can somebody elaborate on that? Thanks.

jwhui, 2019-08-08:
@bstraehl, thanks for raising this issue.

I believe the 2x factor comes from the fact that many dev boards include USB-to-serial converters that support two channels. I agree that the current mapping from node id to dev path is not ideal.

Couple possibilities that immediately come to mind:

1. Add another parameter that can specify whether to apply the 2x factor.

2. Have a config file that maps between node id and device path.

Open to other ideas.

bstraehl, 2019-08-12:
@jwhui Maybe another possibility would be to have a command line option allowing to enumerate interfaces. In my own tools I usually implement something like this:
--another-option --interface /dev/ttyUSB0,/dev/ttyUSB1,/dev/ttyUSB2 --another-option
```
    parser.add_argument("--interface", "-i", dest="interface", action="append", nargs="+", required=True,
                        help="One or more serial interface(s) to use to communicate with the module(s). Please specify comma separated")
```
```
    serialInterfaces = []
    for interfaceOptionSpec in args.interface:
        for interfaces in interfaceOptionSpec:
            for i in interfaces.split(","):
                serialInterfaces.append(serial.Serial(i, baudrate=args.bitrate, rtscts=args.rtscts, timeout=0.0001))
```
And having the option to load a configuration file is always a good idea I think.

bstraehl, 2019-08-12:
It's also necessary to tweak Node.destroy in order to run certification tests smoothly on SOCs. I had to change the method as follows:
```
@@ -161,10 +161,11 @@ class Node:
             return
 
         if (
-            hasattr(self.pexpect, 'proc')
-            and self.pexpect.proc.poll() is None
-            or not hasattr(self.pexpect, 'proc')
-            and self.pexpect.isalive()
+            (hasattr(self.pexpect, 'proc')
+             and self.pexpect.proc.poll() is None
+             or not hasattr(self.pexpect, 'proc')
+             and self.pexpect.isalive()) 
+            and not (self.node_type == 'soc')
         ):
             print("%d: exit" % self.nodeid)
             self.pexpect.send('exit\n')
```
Adding a check whether we are running on SOC-mode helps to prevent the tear down of the simulation processes, which we only want to do on PC and not on SOCs. This is for sure not a clean solution, but it helps. Can it be that I'm doing something wrong, or are Thread certification tests seldomly ran on real targets?

jwhui, 2019-09-05:
@bstraehl, thanks for this update. As you surmised, we primarily use the python test scripts for Travis CI tests and not on real targets.

jwhui, 2019-09-17:
@bstraehl, would you like to submit a PR with your enhancements above?

jwhui, 2019-11-14:
Closing stale issue.

closed 2019-11-14 by jwhui

bstraehl, 2019-11-15:
@jwhui 
Sorry, for letting this getting stale. I considered multiple times to do a PR, but my changes are not fully aligned with the concepts of this certification test environment and are not mature enough. They just helped me to track down issues such as https://groups.google.com/forum/?hl=en-GB#!topic/openthread-users/JAvrPZh2Sao. It's not very satisfying to not have the time to do things right, but maybe I'll find the time to do it properly.
Thank you for your very valuable and highly estimated work.

--
Beat Straehl

jwhui, 2019-11-15:
@bstraehl, thanks for the note! Completely understand. If you do happen to find the time in the future, contributions are always welcome!